slic3r/Slic3r #4015: Slic3r 1.3.0 does not run on openSuSE 13.2 x64

azalanono, 2017-06-07:
### Version
1.3.0

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
openSUSE 13.2 x64 with all updates done (as long as there were ones.

### Behavior
At startup I only get this:
rainer@majestix:~/Slic3r> ./Slic3r --gui
Can't load '/home/rainer/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /usr/lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /home/rainer/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so) at DynaLoader.pm line 193.
 at /home/rainer/Slic3r/local-lib/lib/perl5/Slic3r.pm line 53.
Compilation failed in require at /home/rainer/Slic3r/local-lib/lib/perl5/Slic3r.pm line 53.
BEGIN failed--compilation aborted at /home/rainer/Slic3r/local-lib/lib/perl5/Slic3r.pm line 53.
Compilation failed in require at ./slic3r.pl line 16.
BEGIN failed--compilation aborted at ./slic3r.pl line 16.

It would be helpful to take more care on backwards compatibility to older systems.

lordofhyphens, 2017-06-07:
The oldest compiler we support is gcc 4.9.

We cannot and will not support older.

The builds are done on Ubuntu 14.04 LTS systems with gcc4.9 installed.

lordofhyphens, 2017-08-22:
@azalanono libstdc++ should now be statically linked (as of June 7th).

azalanono, 2017-08-22:
Hello.   "LoH"  (on  #slic3r) pleased to report issues with 1.0.3-dev here.  So I do.

My system is openSUSE 13.2 x64.

I downloaded right now the latest version "Slic3r-1.3.0-dev-a54ca92-x86_64.AppImage" from slic3r.org, made it executable, and tried to start it from a terminal.

First try:
./Slic3r-1.3.0-dev-a54ca92-x86_64.AppImage --version
1.3.0-dev

Trying to fire it up:
./Slic3r-1.3.0-dev-a54ca92-x86_64.AppImage --gui 
Can't load '/tmp/.mount_Slic3ritovlH/usr/lib/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so' for module Wx: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /tmp/.mount_Slic3ritovlH/usr/lib/libwx_gtk2u_adv-3.0.so.0) at DynaLoader.pm line 193.
 at /tmp/.mount_Slic3ritovlH/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at /tmp/.mount_Slic3ritovlH/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3ritovlH/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at (eval 84) line 1.

I hope this is helpful to you.

lordofhyphens, 2017-08-23:
@azalanono That should be sorted out now in the AppImage build.

lordofhyphens, 2017-09-02:
@azalanono Closing again.

closed 2017-09-02 by lordofhyphens

GWRon, 2017-11-09:
```
$ ./Slic3r-1.37.2-prusa3d-linux64-full-201710211407.AppImage 
Can't load '/tmp/.mount_Slic3rQkz9Gj/usr/bin/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /tmp/.mount_Slic3rQkz9Gj/usr/bin/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so) at /tmp/.mount_Slic3rQkz9Gj/usr/bin/lib/site_perl/5.22.0/XSLoader.pm line 70.
 at /tmp/.mount_Slic3rQkz9Gj/usr/bin/lib/site_perl/5.22.0/Slic3r/XS.pm line 22.
Compilation failed in require at /tmp/.mount_Slic3rQkz9Gj/usr/bin/lib/site_perl/5.22.0/Slic3r.pm line 58.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rQkz9Gj/usr/bin/lib/site_perl/5.22.0/Slic3r.pm line 58.
Compilation failed in require at /tmp/.mount_Slic3rQkz9Gj/usr/bin/bin/slic3r.pl line 16.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rQkz9Gj/usr/bin/bin/slic3r.pl line 16.
```

For me it still fails with the .appImage file.

lordofhyphens, 2017-11-09:
@gwron that isn't our Slic3r.

Https://dl.slic3r.org/dev/linux

GWRon, 2017-11-09:
Ahh sorry, my mistake. Just googled for the error, found the similar issue and missed that "little" detail.

Thanks for clarifying.

lordofhyphens, 2017-11-10:
Since you're on that version of OpenSuse, can you try the main appimage
from dl.slic3r.org?

PR #4016 (2017-06-09): Link to libstdc++ statically.
Attempting to link to libstdc++ statically to deal with relatively ancient systems (see: opensuse 13.2) which are never getting a more modern compiler than GCC 4.8.

@azalanono Fixes #4015