slic3r/Slic3r #4017: Infill Overlap Causing Print Failures on Bottom Layers

lordofhyphens, 2017-06-08:
### Version 
1.2.9, also 1.3.0-dev as of 9f0335b 

### Operating system type + version
Windows 8

### Behavior
With the provided configuration and changing the infill/perimeter overlap to 15%, I got bottom layers that were 1.2mm wide and 0.60mm thick using 0.6mm extrude and 0.30mm first layers on a spiral fill circular area. The nozzle then dragged itself through the first layer crossing to the other side of the object. This occurred repeatedly using 15% infill Overlap and was avoided when infill/perimeter overlap was set to 0%.

[image]

#### STL/Config (.ZIP) where problem occurs
https://github.com/alexrj/Slic3r/files/1059803/4012.zip

(Submitted for @JanComputerman as the result of splitting #4012 into different issues. @JanComputerman please subscribe to this issue)

JanComputerman, 2017-06-09:
The latest version 1.3-DEV seems to work ok but this was an interesting problem and 1.2mm wide 0.60mm thick extrudes from a 0.40mm nozzle was high volume printing. The direct drive stepper was starting to skip a little!

lordofhyphens, 2017-06-09:
You should not be exceeding the nozzle diameter for layer height.

JanComputerman, 2017-06-10:
Yes it will not let you even if you try ... nozzle is 0.40mm, layer 0.31mm, the max I use rarely is 0.35mm layer usually 0.25-0.30mm but interesting that 0.31mm stopped the center gap on the first layer.
--------------------------------------------

 

 
 
 
 You are
 receiving this because you were mentioned.
 it on GitHub, or mute
 the thread.

lordofhyphens, 2017-06-10:
Auto width or fixed width?

JanComputerman, 2017-06-10:
All widths set to 0.50mm and infill overlap settings of 0-10-15-50 had no effect. This is a 66x71mm square area with straight sides object not the object shown above but really showing some unexpected effect. Trying other first layer 0.20 0.21 0.23 0.26 0.29 0.31 0.32 0.34 no gap, 0.22 0.24 0.25 0.27 0.28 0.30 0.33 gap.