slic3r/Slic3r #4018: Poor Bridging Infill Anchor Points at Start/Stop of Path

lordofhyphens, 2017-06-08:
### Version 
1.2.9, also 1.3.0-dev as of 9f0335b 

### Operating system type + version
Windows 8

### Behavior
I will say that the infill Overlap setting greatly improves bridging when connecting to the vertical walls but there seems to be a gap problem when the walls are not parallel to the X or Y axis where the bridge doesn't reach the wall at all.

[image]

The anchor points are not very effective for the beginning and end of bridging infill at 0% infill.

#### STL/Config (.ZIP) where problem occurs
https://github.com/alexrj/Slic3r/files/1059803/4012.zip

(Submitted for @JanComputerman as the result of splitting #4012 into different issues. @JanComputerman please subscribe to this issue)

JanComputerman, 2017-06-08:
All extrudes at 0.5mm 50% infill overlap and 1.3.dev slic3r which crashes when trying to connect to my PrintrBot Jr so I exported the gcode and here is that infamous layer ...
[image]

JanComputerman, 2017-06-09:
The edge where it fails to connect is at an angle to the x axis and also at 45 degrees from vertical 3 x 0.5mm perimeters wide on top so it is also possible that the "vertical" wall may not be exactly where it is supposed to be causing the failure to connect. The layer is printed right to left and joins just fine as the narrow section is reached. It fails again as it reaches the other end with the same shaped opening.

dobrzann, 2017-06-17:
I have the same problem.
nozzle 0.4mm
layer 0.2mm
[image]
[image]
[image]
[image]

Model:
[50x50 trojkat.zip](https://github.com/alexrj/Slic3r/files/1082501/50x50.trojkat.zip)

dobrzann, 2017-06-17:
It looks better after I changed OVERLAP_FACTOR in flow.hpp form 1 to 2. 
Of course except walls and other perimeters
[image]

After that change infill looks better too.

JanComputerman, 2017-06-19:
Yeah that lower green image is classic zigzag fill no where near the wall like you have play in your drive and it is not going all the way to the wall ... seen that when it tried to fill a long thin gap and the extruder just buzzes in the Y direction as it slowly moves the X down the slot gap. Is there a gcode editor that could let you select a number of points and just move them a little using a graphical interface? Then we could make little adjustments to these little gap problems.

JanComputerman, 2017-07-06:
Ok I tried the 6/24 release of Slic3r and it now seems to default to 55% Overlap and if set to 50% will drop the first 2 or 3 bridges at the narrow end where it begins to print the top layer. Another odd thing however is that it prints the layers (2 perimeters) at a modest speed until the layer prior to bridging and then will print that perimeter at a very high speed guessing 60mm/second ... it just whips around the top then bridges.