slic3r/Slic3r #4019: Missing leading zero in [hour], [minute] and [second]

foreachthing, 2017-06-08:
### Version
slic3r-gui3-shortcuts.2017.05.26.1123.a1b0246.64bit

### Operating system type + version
Windows 7 Pro

### Behavior
* instead of [timestamp] (where it works), I use [hour]. If [hour] < 10, then there is no leading zero. Same for minutes and seconds. This could lead to this time: 913 instead of 090103.

lordofhyphens, 2017-06-09:
[timestamp] has the special processing that you are looking for, the entire point of the separated ones is to give you the exact number without any special processing.

See: 
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/PlaceholderParser.cpp, function process(). [timestamp] should be valid anywhere [hour], etc are supported.

(unrelated: you should probably stop using gui3-shortcuts, it isn't being maintained as it was merged).

foreachthing, 2017-06-09:
> stop using gui3-shortcuts

Thanks. I'll try :-)
Then I'll fudge something together with my post processing... I can add anything to my new file name right there. Sorry to bother you with this. - closed -

lordofhyphens, 2017-06-09:
It would help if you explained the use case better--why you wanted the output to be that way.

It isn't hard at all to have separate date and time entries (you can see what goes into the time stamp right there), but I would like to put some thought into the interface as well.

foreachthing, 2017-06-09:
The Ultimaker 2 has a limited filename length display capability. So, I wanted to make sure I print the latest file (without cleaning the SD-card all the time). I think that 090103 makes a more human readable time than 913.

alexrj, 2017-06-09:
Zero-padding the individual date/time variables makes sense and I can't see what could go wrong with doing it. @lordofhyphens what do you think?

lordofhyphens, 2017-06-09:
I originally thought so too, I was mostly concerned with getting another
request to change it back.

We can deal with that when it happens, I guess.

closed 2017-06-09 by lordofhyphens

lordofhyphens, 2017-06-09:
@foreachthing There you go. 
@alexrj Refactored things a bit as well to re-use the variables that got set in the timestamp generation.

foreachthing, 2017-06-09:
Thanks a lot! :-)

commit 5e7fbe17 (2017-06-09): Always format month/day/hour/second/minute placeholder variables out to their full width.

Fixes #4019