slic3r/Slic3r #4021: Crash in applying start gcode with '\' at end of line.

lordofhyphens, 2017-06-09:
### Version
5e7fbe1

### Operating system type + version
Windows 10

### Behavior
Entering a `\` at the end of a line for custom start gcode causes Slic3r to close/crash abruptly. 

````
Error in function apply_only at src\libslic3r\Config.cpp:267: Unexpected failure when deserializing serialized value for start_gcode
         at C:/Users/lenox/Slic3r/lib/Slic3r/GUI/PresetEditor.pm line 152.
        Slic3r::GUI::PresetEditor::__ANON__() called at C:/Users/lenox/Slic3r/lib/Slic3r/GUI.pm line 207
        Slic3r::GUI::__ANON__(Slic3r::GUI::MainFrame=HASH(0x6a9ea08), Wx::IdleEvent=SCALAR(0x11e85700)) called at .\slic3r.pl line 125
        eval {...} called at .\slic3r.pl line 1251
````

### Reproduction Steps
1. Start Slic3r
    * To get error message on Windows, start with the debug console or invoke `perl ./slic3r.pl`
1. Open Printer Settings preset editor
1. Switch to custom gcode editor
1. At the end of a line, enter in a single `\`
    * Inserting a `\` before a space at the end of the line does not cause a crash.

JanComputerman, 2017-06-09:
The \ character is used in "C" type languages for indicating that the next text character is to be converted to a control character i.e. "\n" is print a new line when in a text string. You may be triggering a native language function and not a slic3r program function. Comments begin with ; in gcode.

lordofhyphens, 2017-06-09:
Yeah, I know. I didn't have time to run it down and wanted to capture it
for other people to look at.

lordofhyphens, 2017-09-04:
Thanks @tmotl for the patch.

closed 2017-09-04 by lordofhyphens