slic3r/Slic3r #4024: 3D Honeycomb infill does not honor extrusion width

DrLex0, 2017-06-12:
### Version
slic3r-1.3.0-dev-2bedc5c
Also present in Prusa3D fork.

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* The 3D Honeycomb infill always prints at the same extrusion width even though the pattern spacing varies with infill percentage, the result is that the actual infill percentage is incorrect.
* _Steps needed to reproduce the problem_
  * Slice any solid object with any settings, infill set to 3D Honeycomb.
  * In the Advanced settings, set Infill extrusion width to 0.4 and look at the preview
  * Go back to the Advanced settings, set infill extrusion width to 1.0 and look at the preview again
* _Expected Results_
  * The extrusions are much thicker with extrusion width set to 1.0 than 0.4, and the pattern is printed larger to compensate for the wider extrusion.
* _Actual Results_
  * The pattern is printed larger, but the extrusion width is the same instead of wider. It looks like the width is always set to the nozzle diameter. This makes the infill too sparse at wider extrusion settings.

I checked all other infills and 3D Honeycomb is the only one with this problem.

[image]

[image]
Note how the infill is moved further away from the perimeters, which is as expected if the extrusions would really be wider. They just aren't.

gringer, 2017-06-24:
The infill percentage is probably not correct either. The spacing is based on a square infill pattern, and I originally coded it to match the square pattern when the diagonals disappeared. I'm working on an update of the 3D honeycomb infill which should be better supported because it doesn't have any gaps between each layer. See issue #4045 for more details. I'll keep the density calculation in mind when porting the R code to perl (for Slic3r) and C++ (for Cura2).

gringer, 2019-01-04:
Is this still an issue with the updated 3D honeycomb infill pattern (i.e. post Feb-2018)?

DrLex0, 2019-01-04:
@gringer Yes, even in the latest master build (as well as the latest Prusa3D alpha) the extrusion width setting for this infill pattern is not respected.

supermerill, 2019-01-04:
That's because it use the "bridge" flag. So it prints as a bridge.

Imho, it's something to change as this has many culprits.

gringer, 2019-01-20:
Odd. It seemed to work better for me without bridging; I don't think it needs that anymore.

https://github.com/slic3r/Slic3r/pull/4049#issuecomment-312119751