slic3r/Slic3r #4025: OpenGL.c: loadable library and perl binaries are mismatched (got handshake key 0xdb80080, needed 0xde00080)

Fahim-Ferdous, 2017-06-13:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-06-13:
If this is the AUR or the normal package, contact its maintainer. Odds are your version of system perl changed at some point (hence the mismatch). We can't support the Arch package (it's not our work) here.

In the meantime, give the download at http://dl.slic3r.org/dev/linux a try.

closed 2017-06-13 by lordofhyphens

Fahim-Ferdous, 2017-06-13:
Thanks @lordofhyphens  :)

NateTG, 2018-04-07:
I ran into something similar recently that affected slic3r and Build.PL.    It would be nice if there was some kind of 'make clean' equivalent

lordofhyphens, 2018-04-07:
@NateTG delete local-lib dir.