slic3r/Slic3r #4026: Build system rebuilds all files

fivethreeo, 2017-06-13:
Any way to make it rebuild only changed files?

lordofhyphens, 2017-06-13:
If you set up local::lib properly in your shell, you can jump into `xs` and run Build.PL -> `Build`  -> `Build install` there.

On Windows, this would be `Invoke-Expression "$(perl -Mlocal::lib=C:\path\to\Slic3r\local-lib)"`
On Linux/OSX: `eval $(perl -Mlocal::lib=/path/to/slic3r/local-lib`.

You may still need to rerun the main Build.PL file if you run into weirdness, it's not the robust build system around.

closed 2017-06-13 by lordofhyphens

lordofhyphens, 2017-06-13:
You can also run tests with `prove ./t` in the `xs` dir if you want to run the XS tests and in the main dir to run the rest of the tests. 

You do need to have the shell set up properly as above to do this.