slic3r/Slic3r #4027: G-code generation issue: fan activate on 1st layer when using 1 raft layer even if fan disabled for 1st layer

bmx666, 2017-06-13:
### Version
commit https://github.com/alexrj/Slic3r/commit/46504db02f7df3c6ece0dcc1abae38c9231cc61f

### Operating system type + version
Ubuntu 16.04 x64

### Behavior
* 3D-printer **Hercules NEW** - RepRap G-code compatible
* Set raft layers = 1
* Disable fan for 1st layer

### G-code with one raft layer
```
G28 
M106 S0 ; set fan off
M104 S210 
M140 S65 
M190 S65 
M109 S210 
G1 X90 F3000 
G1 Z0 F400

G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion

; 1st layer
G92 E0
M106 S255 ; set fan full speed
G1 Z0.200 F3600.000
G1 E-2.50000 F3600.00000
G92 E0
G1 Z0.600 F3600.000
G1 X50.873 Y49.698 F3600.000
G1 Z0.200 F3600.000
G1 E2.50000 F3600.00000
M204 S9000
G1 F600
G1 X53.065 Y48.020 E2.59966
...
G1 X126.376 Y126.021 E221.24462

; 2nd layer
M104 S205 ; set temperature
G1 Z0.400 F3600.000
G1 X126.867 Y124.524 F3600.000
G1 F1500
G1 X126.665 Y124.671 E221.25364
```

### G-code without raft layer
```
G28 
M106 S0 ; set fan off
M104 S210 
M140 S65 
M190 S65 
M109 S210 
G1 X90 F3000 
G1 Z0 F400

G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion

; 1st layer
G92 E0
G1 Z0.200 F3600.000
G1 E-2.50000 F3600.00000
G92 E0
G1 Z0.600 F3600.000
G1 X50.627 Y51.119 F3600.000
G1 Z0.200 F3600.000
G1 E2.50000 F3600.00000
M204 S9000
G1 F600
G1 X51.797 Y50.110 E2.55580
...
G1 X125.711 Y123.447 E50.08531
G1 Z0.200 F3600.000

; 2nd layer
M106 S255 ; set fan full speed
M104 S205 ; set temperature
G1 Z0.400 F3600.000
```

lordofhyphens, 2017-06-13:
Need the basic information that was requested in the text you deleted.

bmx666, 2017-06-13:
With raft layer:
[image]

Without raft layer:
[image]

Model:
GuitarWallMount.STL from [here](https://www.thingiverse.com/thing:481198)

bmx666, 2017-06-13:
### Print settings INI-file
```
# generated by Slic3r 1.3.0-dev on 2017-06-13 14:45:41
avoid_crossing_perimeters = 1
bottom_infill_pattern = rectilinear
bottom_solid_layers = 2
bridge_acceleration = 0
bridge_flow_ratio = 1
bridge_speed = 15
brim_connections_width = 0
brim_width = 1
compatible_printers = 
complete_objects = 0
default_acceleration = 9000
dont_support_bridges = 0
external_perimeter_extrusion_width = 0.5
external_perimeter_speed = 25
external_perimeters_first = 1
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0.5
fill_angle = 90
fill_density = 20%
fill_gaps = 1
fill_pattern = 3dhoneycomb
first_layer_acceleration = 0
first_layer_extrusion_width = 0.5
first_layer_height = 0.2
first_layer_speed = 50%
gap_fill_speed = 15
gcode_comments = 0
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0.5
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 60
interface_shells = 0
interior_brim_width = 1
layer_height = 0.2
max_print_speed = 90
max_volumetric_speed = 0
min_skirt_length = 0
notes = Режим печати материалом PLA (Bestfilament) с диаметром сопла 0,5 мм;\nПрименяется для печати крупногабаритных моделей; \nВысота слоя - 0.2 мм;\nЗаполнение - 15%;\nСкорость печати периметров  65 мм/сек;\nТемпература экструзии/подогрева стола - 210°/65°.\n
only_retract_when_crossing_perimeters = 0
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0.5
perimeter_speed = 50
perimeters = 3
post_process = 
print_settings_id = 
raft_layers = 1
resolution = 0
seam_position = nearest
shortcuts = support_material
skirt_distance = 10
skirt_height = 1
skirts = 1
small_perimeter_speed = 25
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0.5
solid_infill_speed = 65%
spiral_vase = 0
standby_temperature_delta = -5
support_material = 1
support_material_angle = 45
support_material_buildplate_only = 0
support_material_contact_distance = 0
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0.5
support_material_interface_extruder = 1
support_material_interface_layers = 0
support_material_interface_spacing = 2.5
support_material_interface_speed = 100%
support_material_pattern = rectilinear-grid
support_material_spacing = 0.25
support_material_speed = 20
support_material_threshold = 90%
thin_walls = 1
top_infill_extrusion_width = 0.5
top_infill_pattern = rectilinear
top_solid_infill_speed = 15
top_solid_layers = 3
travel_speed = 60
xy_size_compensation = 0
```

bmx666, 2017-06-13:
### Printer settings INI-file:
```ini
# generated by Slic3r 1.31.6-prusa3d-win64 on Tue Dec 13 12:09:02 2016
bed_shape = 0x0,180x0,180x180,0x180
before_layer_gcode = 
end_gcode = G28 X0 ; home X axis\nG28 Z0; home Z axis\nG0 Y150\nG92 Z0\nM104 S0 ; turn off temperature\nM140 S0 ; turn off temperature\nM84 ; disable motors
extruder_offset = 0x0
gcode_flavor = reprap
layer_gcode = 
nozzle_diameter = 0.5
octoprint_apikey = 
octoprint_host = 
pressure_advance = 0
printer_settings_id = 
retract_before_travel = 2
retract_layer_change = 0
retract_length = 2.5
retract_length_toolchange = 10
retract_lift = 0.4
retract_lift_above = 0
retract_lift_below = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 60
serial_port = 
serial_speed = 250000
start_gcode = G28 \nM106 S0 \nM104 S[first_layer_temperature] \nM140 S[first_layer_bed_temperature] \nM190 S[first_layer_bed_temperature] \nM109 S[first_layer_temperature] \nG1 X90 F3000 \nG1 Z0 F400\n
toolchange_gcode = 
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
wipe = 0
z_offset = 0
```

bmx666, 2017-06-13:
### Print settings INI-file
```ini
# generated by Slic3r 1.3.0-dev on 2017-06-13 16:50:31
avoid_crossing_perimeters = 1
bottom_infill_pattern = rectilinear
bottom_solid_layers = 2
bridge_acceleration = 0
bridge_flow_ratio = 1
bridge_speed = 15
brim_connections_width = 0
brim_width = 0
compatible_printers = 
complete_objects = 0
default_acceleration = 9000
dont_support_bridges = 0
external_perimeter_extrusion_width = 0.5
external_perimeter_speed = 25
external_perimeters_first = 1
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0.5
fill_angle = 90
fill_density = 20%
fill_gaps = 1
fill_pattern = 3dhoneycomb
first_layer_acceleration = 0
first_layer_extrusion_width = 0.5
first_layer_height = 0.2
first_layer_speed = 50%
gap_fill_speed = 15
gcode_comments = 0
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0.5
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 60
interface_shells = 0
interior_brim_width = 0
layer_height = 0.2
max_print_speed = 90
max_volumetric_speed = 0
min_skirt_length = 0
notes = Режим печати материалом PLA (Bestfilament) с диаметром сопла 0,5 мм;\nПрименяется для печати крупногабаритных моделей; \nВысота слоя - 0.2 мм;\nЗаполнение - 15%;\nСкорость печати периметров  65 мм/сек;\nТемпература экструзии/подогрева стола - 210°/65°.\n
only_retract_when_crossing_perimeters = 0
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
overridable = support_material
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0.5
perimeter_speed = 50
perimeters = 3
post_process = 
print_settings_id = 
raft_layers = 1
resolution = 0
seam_position = nearest
skirt_distance = 10
skirt_height = 1
skirts = 1
small_perimeter_speed = 25
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0.5
solid_infill_speed = 65%
spiral_vase = 0
standby_temperature_delta = -5
support_material = 1
support_material_angle = 45
support_material_buildplate_only = 0
support_material_contact_distance = 0
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0.5
support_material_interface_extruder = 1
support_material_interface_layers = 0
support_material_interface_spacing = 2.5
support_material_interface_speed = 100%
support_material_pattern = rectilinear-grid
support_material_spacing = 0.25
support_material_speed = 20
support_material_threshold = 90%
thin_walls = 1
top_infill_extrusion_width = 0.5
top_infill_pattern = rectilinear
top_solid_infill_speed = 15
top_solid_layers = 3
travel_speed = 60
xy_size_compensation = 0
```

bmx666, 2017-06-13:
Stable branch has same issue, but in version 1.2.9 everything is OK under Linux and Windows.

lordofhyphens, 2017-06-13:
@bmx666 as a fyi, you can attach zip files.

lordofhyphens, 2017-06-13:
File->Export Config (Ctrl-E), zip and attach. No extra questions about what is being used.

bmx666, 2017-06-13:
[config.zip](https://github.com/alexrj/Slic3r/files/1072353/config.zip)

bmx666, 2017-06-13:
Update G-code comments.

### Issue
**M106 S255** G-code command is inserted before 1st layer. Even if in settings fan was disabled for n-th layers.

### Fix
Insert **M106 S255** after n-th first layers according to settings

bmx666, 2017-10-13:
The issue is still exists with the latest commit a28f431

closed 2022-02-18 by bmx666