slic3r/Slic3r #4030: Support too complicated (at least for soluble material)

Martronic-SA, 2017-06-16:
### Version
Last dev version

### Operating system type + version
Windows

### Behavior
Rectilinear, Rectilinear grid, hoeycomb support is too complicated and very difficult to print. Too many unuseful details. No contact between vertical shell and infill.

In the sample below:
file is MAZE LAMP [https://www.thingiverse.com/thing:2071943/#files](https://www.thingiverse.com/thing:2071943/#files) reduced to 25%. 0.4 nozzle, 0.2 layer thickness.
[image]

The problem is that the support infill is very difficult to print as it doesnt touch the vertical shell.
Why not just do the same as an usual infill?

[image]

Martronic-SA, 2017-11-29:
Also here very complicated for nothing if you have soluble material:
[image]
This doesn't allow to have a nice under surface.
With a soluble material the support should be treaten exactly the same as standard perimeters/infill.

bubnikv, 2017-11-29:
The Prusa3D fork applies a concentric interface layer by default for the soluble material.

Martronic-SA, 2017-11-29:
Ok, interesting. Will there be a PR for that?

lordofhyphens, 2017-11-29:
The rest of the implementation has to be ported first.