slic3r/Slic3r #4034: Raft + Brim: Brim too late

aligator, 2017-06-18:
### Version
https://github.com/alexrj/Slic3r/commit/abc96ad3bf65aa2eec9a24de3973e02a4130c471
1.2.9-1111-gabc96ad

### Operating system type + version
Linux Mint 18.1 x64

### Behavior
Settings: 
layer height: 0.28
first layer height: 0.28
Brim width: 10
Raft Layers: 3

Result:
the Brim is not printed on the first layer. Instead it is shifted up --> it is in the air

Expected:
the Brim should be on the first Layer

#### STL/Config (.ZIP) where problem occurs
https://www.thingiverse.com/thing:330151
[config.ini.zip](https://github.com/alexrj/Slic3r/files/1083269/config.ini.zip)

lordofhyphens, 2017-06-23:
Some kind of regression going on here, the brim is being printed on the first layer of the model, but offset from the raft's position.

lordofhyphens, 2017-06-23:
It's a regression since after 12165c7.

lordofhyphens, 2017-06-23:
Test case:
* With raft_layers > 0 and brim width > 0, the number of G1 Z moves before the first brim print move must be exactly one.
    * (the initial move to first layer height specifically, but that isn't necessarily germane to this test)

lordofhyphens, 2017-06-23:
Bisect log so far.
````
git bisect start
# bad: [aa20fb65e0103ae6e3cc8e5e3e66aeb7ddbbc5a1] Bugfix: refer to getBool() instead of getting a reference to the object itself for config options. Fixes #4043
git bisect bad aa20fb65e0103ae6e3cc8e5e3e66aeb7ddbbc5a1
# good: [12165c727e57c124e5ac86c57561ff6541772d09] Ported brim generation to C++/XS
git bisect good 12165c727e57c124e5ac86c57561ff6541772d09
# bad: [00744c259cb835af7b3b992a93cd99aad5734f35] Append the --gui flag if compiled with -DFORCE_GUI.
git bisect bad 00744c259cb835af7b3b992a93cd99aad5734f35
# good: [6feaef72edaa8df64be8b78306f29730be62da3f] Merge pull request #3691 from lordofhyphens/max_width_3x_nozzle
git bisect good 6feaef72edaa8df64be8b78306f29730be62da3f
# skip: [4c14c656f5c9adf712dfbcea0dc1b7c044131c37] Updated MANIFEST
git bisect skip 4c14c656f5c9adf712dfbcea0dc1b7c044131c37
# bad: [488cc02f531d63e8d87a3a572853a391ae8a7647] Try to fix OctoPrint issues. #3789
git bisect bad 488cc02f531d63e8d87a3a572853a391ae8a7647
# skip: [abbc458f106c655c487eec0937e01add02373be5] Fix stats output
git bisect skip abbc458f106c655c487eec0937e01add02373be5
# bad: [ecc171eca6971922441bf0d713ee400582373e47] Bugfix: use_relative_e_distances and spiral vase were broken again after the port. #3784
git bisect bad ecc171eca6971922441bf0d713ee400582373e47
# good: [6b334e29dcf5a6fafab82b5228eba95b5db38dd8] change public link for auto builds
git bisect good 6b334e29dcf5a6fafab82b5228eba95b5db38dd8
````

lordofhyphens, 2017-06-24:
9dbe2eea37eb2b6a219bff7c12852df25d07bee8 is the first bad commit
commit 9dbe2eea37eb2b6a219bff7c12852df25d07bee8
Author: Alessandro Ranellucci <aar@cpan.org>
Date:   Mon Mar 20 15:46:22 2017 +0100

    Menu option for coloring toolpaths by extruder using the configured filament colors

:040000 040000 9012fe2eeef955ec01f491ff1b1da8469ea556ca 3e6fa3b9cb7bba922e899d3f7a1c3ab6950028dd M	lib
:040000 040000 3822b34a635294bc9cd7c345c1f21f53df5d8aa6 1b66e40b31107482b506ddd5ca0b1d91989a253f M	xs

lordofhyphens, 2017-06-24:
If someone wants to look further, here's my bisect log.

git bisect start
# bad: [aa20fb65e0103ae6e3cc8e5e3e66aeb7ddbbc5a1] Bugfix: refer to getBool() instead of getting a reference to the object itself for config options. Fixes #4043
git bisect bad aa20fb65e0103ae6e3cc8e5e3e66aeb7ddbbc5a1
# good: [12165c727e57c124e5ac86c57561ff6541772d09] Ported brim generation to C++/XS
git bisect good 12165c727e57c124e5ac86c57561ff6541772d09
# bad: [00744c259cb835af7b3b992a93cd99aad5734f35] Append the --gui flag if compiled with -DFORCE_GUI.
git bisect bad 00744c259cb835af7b3b992a93cd99aad5734f35
# good: [6feaef72edaa8df64be8b78306f29730be62da3f] Merge pull request #3691 from lordofhyphens/max_width_3x_nozzle
git bisect good 6feaef72edaa8df64be8b78306f29730be62da3f
# skip: [4c14c656f5c9adf712dfbcea0dc1b7c044131c37] Updated MANIFEST
git bisect skip 4c14c656f5c9adf712dfbcea0dc1b7c044131c37
# bad: [488cc02f531d63e8d87a3a572853a391ae8a7647] Try to fix OctoPrint issues. #3789
git bisect bad 488cc02f531d63e8d87a3a572853a391ae8a7647
# skip: [abbc458f106c655c487eec0937e01add02373be5] Fix stats output
git bisect skip abbc458f106c655c487eec0937e01add02373be5
# bad: [ecc171eca6971922441bf0d713ee400582373e47] Bugfix: use_relative_e_distances and spiral vase were broken again after the port. #3784
git bisect bad ecc171eca6971922441bf0d713ee400582373e47
# good: [6b334e29dcf5a6fafab82b5228eba95b5db38dd8] change public link for auto builds
git bisect good 6b334e29dcf5a6fafab82b5228eba95b5db38dd8
# good: [72681aca01a108898666f8c71d655c93230c8a9a] Don't perform CPAN tests for IO::Scalar. #3573
git bisect good 72681aca01a108898666f8c71d655c93230c8a9a
# bad: [6d0cb54cf2922c1a5e0d342462298ab1e54c854c] A few more fixes to plater
git bisect bad 6d0cb54cf2922c1a5e0d342462298ab1e54c854c
# good: [dddbe64b866e10a7c88645faadb77a2508341d1e] Restrict file types for model exports to what we are permitting (UX fix). Changed extension from .amf.xml -> .amf (common usage) Fixes #3774
git bisect good dddbe64b866e10a7c88645faadb77a2508341d1e
# good: [a19cc10b8e7f126a3f0f102716bb4d8536f3590f] Bugfix: Controller was crashing on wxWidgets 2.8.x. #3159
git bisect good a19cc10b8e7f126a3f0f102716bb4d8536f3590f
# bad: [9dbe2eea37eb2b6a219bff7c12852df25d07bee8] Menu option for coloring toolpaths by extruder using the configured filament colors
git bisect bad 9dbe2eea37eb2b6a219bff7c12852df25d07bee8
# first bad commit: [9dbe2eea37eb2b6a219bff7c12852df25d07bee8] Menu option for coloring toolpaths by extruder using the configured filament colors

lordofhyphens, 2017-09-03:
@aligator can you retest with current dev build? I've been reviewing the gcode output and it looks like it might just be a graphical glitch (that is, the print is fine). 
turn on gcode comments and you can see where the brim is printed.

lordofhyphens, 2017-09-03:
Alternatively, give it a print ;).

aligator, 2017-09-04:
You are right. It prints correctly.