slic3r/Slic3r #4036: Compilation fails on Ubuntu 14.04

probonopd, 2017-06-19:
Compilation fails on Ubuntu 14.04:

```
sudo apt-get update
sudo apt-get -y install git build-essential libgtk2.0-dev libwxgtk2.8-dev libwx-perl libmodule-build-perl libnet-dbus-perl libexpat1-dev cpanminus liblocal-lib-perl libxmu-dev freeglut3-dev libwxgtk-media2.8-dev libboost-thread-dev libboost-system-dev libboost-filesystem-dev libboost-all-dev libtbb-dev
git clone https://github.com/prusa3d/Slic3r.git
cd Slic3r
export LDLOADLIBS=-lstdc++
export LANGUAGE=en
export LC_ALL=C
sudo perl Build.PL --force

(...)

/usr/include/wx-2.8/wx/string.h:692:3: note:   no known conversion for argument 1 from ‘const char*’ to ‘wxChar {aka wchar_t}’
/usr/include/wx-2.8/wx/string.h:682:3: note: wxString::wxString(int) <near match>
   wxString(int);
   ^
/usr/include/wx-2.8/wx/string.h:682:3: note:   no known conversion for argument 1 from ‘const char*’ to ‘int’
In file included from src/libslic3r/Format/PRUS.cpp:6:0:
/usr/include/wx-2.8/wx/wfstream.h:135:5: error:   initializing argument 1 of ‘wxFFileInputStream::wxFFileInputStream(const wxString&, const wxChar*)’
     wxFFileInputStream(const wxString& fileName, const wxChar *mode = wxT("rb"));
     ^
src/libslic3r/Format/PRUS.cpp:128:21: error: invalid conversion from ‘const char*’ to ‘wxChar {aka wchar_t}’ [-fpermissive]
         if (name == "scene.xml") {
                     ^
In file included from src/libslic3r/Format/PRUS.cpp:5:0:
/usr/include/wx-2.8/wx/string.h:1634:13: error:   initializing argument 2 of ‘bool operator==(const wxString&, wxChar)’ [-fpermissive]
 inline bool operator==(const wxString& s, wxChar c) { return s.IsSameAs(c); }
             ^
src/libslic3r/Format/PRUS.cpp:148:38: error: no matching function for call to ‘wxString::EndsWith(const char [5])’
         else if (name.EndsWith(".stl") || name.EndsWith(".STL")) {
                                      ^
src/libslic3r/Format/PRUS.cpp:148:38: note: candidate is:
In file included from src/libslic3r/Format/PRUS.cpp:5:0:
/usr/include/wx-2.8/wx/string.h:1126:8: note: bool wxString::EndsWith(const wxChar*, wxString*) const
   bool EndsWith(const wxChar *suffix, wxString *rest = NULL) const;
        ^
/usr/include/wx-2.8/wx/string.h:1126:8: note:   no known conversion for argument 1 from ‘const char [5]’ to ‘const wxChar* {aka const wchar_t*}’
src/libslic3r/Format/PRUS.cpp:148:63: error: no matching function for call to ‘wxString::EndsWith(const char [5])’
         else if (name.EndsWith(".stl") || name.EndsWith(".STL")) {
                                                               ^
src/libslic3r/Format/PRUS.cpp:148:63: note: candidate is:
In file included from src/libslic3r/Format/PRUS.cpp:5:0:
/usr/include/wx-2.8/wx/string.h:1126:8: note: bool wxString::EndsWith(const wxChar*, wxString*) const
   bool EndsWith(const wxChar *suffix, wxString *rest = NULL) const;
        ^
/usr/include/wx-2.8/wx/string.h:1126:8: note:   no known conversion for argument 1 from ‘const char [5]’ to ‘const wxChar* {aka const wchar_t*}’
src/libslic3r/Format/PRUS.cpp:150:19: error: ‘wxScopedCharBuffer’ does not name a type
             const wxScopedCharBuffer name_utf8 = name.ToUTF8();
                   ^
src/libslic3r/Format/PRUS.cpp:152:60: error: ‘name_utf8’ was not declared in this scope
             sprintf(model_name_tag, "<model name=\"%s\">", name_utf8.data());
                                                            ^
src/libslic3r/Format/PRUS.cpp:219:40: error: ‘class wxZipInputStream’ has no member named ‘ReadAll’
       if (header.nTriangles > 0 && zip.ReadAll((void*)stl.facet_start, 50 * header.nTriangles)) {
                                        ^
error building src/libslic3r/Format/PRUS.o from 'src/libslic3r/Format/PRUS.cpp' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /home/me/.cpanm/work/1497891152.10765/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
```

lordofhyphens, 2017-06-19:
Some sanity checks, as our build server uses 14.04:

* Is gcc-4.9 (or newer) installed? Slic3r **will not compile** on the default compiler gcc-4.8. This **will not change**. Prusa3d's version doesn't build with gcc-4.8 either. 
* The Wx version we build against is 3.0.2, not 2.8. I don't know how compatible the two are in our environment.
* Also, you're trying to build the Prusa3d fork, we don't support that at all. Try `git clone https://github.com/alexrj/Slic3r.git` instead.

lordofhyphens, 2017-06-19:
https://travis-ci.org/alexrj/Slic3r/builds/244443807

lordofhyphens, 2017-06-19:
If you can get `git clone https://github.com/alexrj/Slic3r.git` to build on your system, then whatever other problems you have should go over to the Prusa3d page.

probonopd, 2017-06-19:
Thanks for your hints, retried with `gcc-4.9` (it was mentioned in the wiki text but not in the apt-get commands for Ubuntu 14.04; added it there) and `https://github.com/alexrj/Slic3r.git`. Works.

closed 2017-06-19 by probonopd

lordofhyphens, 2017-06-19:
Thank you for the wiki edit.

lordofhyphens, 2017-06-19:
As a FYI, the build server output should work on your system.

lordofhyphens, 2017-06-21:
@probonopd are you working on an AppImageKit recipe for Slic3r?

probonopd, 2017-06-22:
Looks like I don't have to, since you have already mastered it 👍 http://dl.slic3r.org/dev/linux/

lordofhyphens, 2017-06-29:
I don't know about "mastered" it; it looks like it works anyway. ^.^; 

At least the way they are being produced on our build server it shouldn't be too much work.