slic3r/Slic3r #4038: Slic3r dropped my settings

gege2b, 2017-06-21:
### Version

Slic3r-master.2017.06.04.1158.9f0335b.64bit

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_

I got this problem a few time the last few days 
Unfortunately, I don't know how to reliably reproduce this. It happens, that's all...

it happens right after I change a shortcut (this time, I changed the infill%) and BOOM, my "prusa slow.ini" is gone and I see the dialogs showed in the screenshot

But most of the time it "just work" like expected. That's why I never been able to reproduce it : I've never managed to find what conditions leads to this bug

I have my settings in a dropbox folder (since some years now, and I never faced any problem until the last week) so I can share the same settings on all my computers

I'm sorry I can't give more informations, but feel free to ask me to test whatever you want

* _Expected Results_

Keep my .ini file alone

* _Actual Results_

Error fired and .ini file deleted
Note : it only deletes the current print settings, but I GUESS it would delete printer or filament one if the problem appear when I change one of their setting

  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

#### STL/Config (.ZIP) where problem occurs
Model is not relevant (I got this problem with a dozen of model)
[Config is here](https://github.com/alexrj/Slic3r/files/1091121/2017-06-21_1128_config.zip)

JanComputerman, 2017-06-21:
I have seen similar weird behavior... I exported configs and imported all 62 if them looked great but since the communication to my USB printer only connects once per power up of the printer I use Repetier and that's when settings for Print, Filament and Printer don't quite get transferred to slic3r 1.3.0-dev  and I have to keep selecting configs or changing setting in what seems to be a default slic3r_settings.ini for Print, Filament and Printer. They work but I have to constantly either select the config or the settings over and over again each time I slice.

lordofhyphens, 2017-06-22:
I doubt that it's actually Slic3r killing your ini file, but some weird interaction with the write handle and Dropbox's daemon.

gege2b, 2017-06-22:
I did some further tests and it seem that indeed it's a conflit with Dropbox, if I close him, all seems fine 

I never had any problem since almost 4 years, the problem appeared when the gui3-shortcuts branch was merged, and only on Windows 10 (it's the computer where I use slic3r the most for my everyday prints)

as slic3r saves the config file as soon as I change a shortcut, dropbox uploads it
Maybe the problem happens when I change another setting BEFORE Dropbox has finished his upload (hard to test)

One solution : handle file writing differently (locking the file and/or looking if it's open before writing to it)
Pro : it would not change the "save on the fly" behaviour for the shortcuts
Con : I don't really know if it's easy/possible

Another solution : Only write the file once (on Slic3r's exit)
Pro : As long as I remember, it's the way slic3r saved their config pre-shortcuts and it worked fine for 4 years :P
Cons : potential lost of shortcuts settings in case of a crash (but it's still better than loosing ALL the config)

Yet another solution : Write settings (if needed) every XX seconds, so cloud softwares could open/upload/close the file
Pro : it's a compromise between "on the fly" and "on exit", reduce the potential lost of settings
Con : far from perfect, could be overkill, and will add some overhead

lordofhyphens, 2017-06-22:
Agreed it needs to be sorted out. I myself keep my settings in a git repository ;)

@alexrj your thoughts?

gege2b, 2017-06-26:
hey, could it be a good idea to add git support into slic3r for settings history ?

something pretty basic like pulling on launch and pushing on exit/setting change would be sufficient I think

lordofhyphens, 2017-06-26:
No. This has come up before, and it is both a lot of work and unrelated
code, and git already has a decent UI and workflow.