slic3r/Slic3r #4040: Behavior when printer expected line number and Slic3r controller line number mismatch is undefined

lordofhyphens, 2017-06-22:
### Version
_Version of Slic3r used goes here_
1.3.0-dev b6708e8

### Operating system type + version
Linux (Debian 8)

### Behavior
If Slic3r is connected to a printer via its Controller and then closed without disconnecting (either through an unrelated crash or a hard-close), re-connecting causes Slic3r and the printer to not agree on what line number it is, so Slic3r tries to resend a value it doesn't have. This will last until the number of sent items by Slic3r reaches that value (usually through its 

One of the possible ways out is to send `M110 0` to the printer to reset its expected  line number, but this is contingent on the current firmware flavor (Machinekit doesn't support it), so GCodeSender will need to be aware of the gcode flavor config options. 

Another is if we don't have a line number is to parse the response to get what the printer is looking for and update our local line number to match what is being asked for so that things match up later. 

Reference: http://reprap.org/wiki/G-code#M110:_Set_Current_Line_Number

@alexrj any preferences on this?

lordofhyphens, 2017-06-22:
This used to cause a segfault in Slic3r (see #4009)

cdedwards, 2017-07-09:
why not have an option to send a specific set of gcode's to the printer when it connects. I get this issue constantly and having the ability to send M110 0 when I connect to the printer along with any other needed g code for setup would be good.

lordofhyphens, 2017-07-09:
I have considered it. Part of it is that such a gcode differs depending on
the firmware. Solvable but there are some problems floating about.

cdedwards, 2017-07-09:
http://prntscr.com/ftlsjj

here's another issue I've run into along this same vein.. suddenly it misplaces a line number. this is the 1.3-dev version. if I M110 and replace the line number, it will continue printing. pretty sure it's not where it left off though

lordofhyphens, 2017-07-09:
@cdedwards It is probably unrelated  to this issue, please open a new one.

lordofhyphens, 2017-07-09:
@cdedwards  Also, you can attach pictures directly to issues, no need to upload to a hosting site.

cdedwards, 2017-07-09:
problem was that it was scrolling after I restarted the printer and it will not stay put. so I hit print screen in order to halt the scroll. guess I could have saved to a file.

lordofhyphens, 2017-07-09:
You can ctrl-V pictures too.