slic3r/Slic3r #4042: 1.3.0-dev-435fa6f-x86_64.AppImage does not launch GUI

probonopd, 2017-06-22:
### Version

Slic3r-1.3.0-dev-435fa6f-x86_64.AppImage

### Operating system type + version

```
linux@linux:~> cat /etc/*release
NAME="openSUSE Leap"
VERSION="42.2"
ID=opensuse
ID_LIKE="suse"
VERSION_ID="42.2"
PRETTY_NAME="openSUSE Leap 42.2"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:leap:42.2"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
```

### Behavior

```
linux@linux:~> /home/linux/Downloads/Slic3r-1.3.0-dev-435fa6f-x86_64.AppImage --gui

(slic3r.pl:7402): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:7402): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
Parentheses missing around "my" list at /tmp/.mount_3AAQ3O/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater.pm line 1870.
Can't load '/tmp/.mount_3AAQ3O/usr/lib/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/OpenGL/OpenGL.so' for module OpenGL: libglut.so.3: cannot open shared object file: No such file or directory at DynaLoader.pm line 193.
 at /tmp/.mount_3AAQ3O/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_3AAQ3O/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
BEGIN failed--compilation aborted at /tmp/.mount_3AAQ3O/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_3AAQ3O/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
BEGIN failed--compilation aborted at /tmp/.mount_3AAQ3O/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
Compilation failed in require at (eval 84) line 1.
```

Possibly `libglut.so.3` needs to be bundled in the AppImage.

lordofhyphens, 2017-06-22:
Probably. I will add that and hope it doesn't break anything.

I think I will add it for now to just the appimage.

lordofhyphens, 2017-06-22:
Actually we'll add it to both because the .tar.bz2 would die too under these conditions.

lordofhyphens, 2017-06-22:
This should be addressed in 837be65

probonopd, 2017-07-10:
Still getting this with `Slic3r-1.3.0-dev-837be65-x86_64.AppImage` on openSUSE LEAP 42.2:

```
Parentheses missing around "my" list at /tmp/.mount_pikzb7/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater.pm line 1870.
Can't load '/tmp/.mount_pikzb7/usr/lib/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/OpenGL/OpenGL.so' for module OpenGL: libglut.so.3: cannot open shared object file: No such file or directory at DynaLoader.pm line 193.
 at /tmp/.mount_pikzb7/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_pikzb7/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
BEGIN failed--compilation aborted at /tmp/.mount_pikzb7/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_pikzb7/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
BEGIN failed--compilation aborted at /tmp/.mount_pikzb7/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
Compilation failed in require at (eval 84) line 1.
```

lordofhyphens, 2017-07-10:
Hrm, missing libglut or missing dependency of libglut?

lordofhyphens, 2017-07-10:
More likely missing dependency of libglut.

lordofhyphens, 2017-07-10:
What about https://dl.slic3r.org/dev/linux/Slic3r-1.3.0-dev-13f121e-x86_64.AppImage ?

probonopd, 2017-07-10:
Also `libglut.so.3: cannot open shared object file: No such file or directory`:

```
linux@linux:~> /home/linux/Downloads/Slic3r-1.3.0-dev-13f121e-x86_64.AppImage  --gui

(slic3r.pl:7690): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:7690): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
Can't load '/tmp/.mount_Slic3rvFfRtR/usr/lib/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/OpenGL/OpenGL.so' for module OpenGL: libglut.so.3: cannot open shared object file: No such file or directory at DynaLoader.pm line 193.
 at /tmp/.mount_Slic3rvFfRtR/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_Slic3rvFfRtR/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rvFfRtR/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_Slic3rvFfRtR/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rvFfRtR/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
Compilation failed in require at (eval 84) line 1.
```

lordofhyphens, 2017-07-10:
Odd, it should be adding libglut.so.3 to the list.
https://github.com/alexrj/Slic3r/blob/master/package/linux/libpaths.txt#L11

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-09-02:
@probonopd Still getting issues?

probonopd, 2017-09-02:
Yes:

```
me@host:~$ Downloads/Slic3r-1.3.0-dev-13f121e-x86_64.AppImage --gui
Can't load '/tmp/.mount_Slic3riOmLnh/usr/lib/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/OpenGL/OpenGL.so' for module OpenGL: libglut.so.3: cannot open shared object file: No such file or directory at DynaLoader.pm line 193.
 at /tmp/.mount_Slic3riOmLnh/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_Slic3riOmLnh/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3riOmLnh/usr/lib/local-lib/lib/perl5/Slic3r/GUI/Plater/2DToolpaths.pm line 126.
Compilation failed in require at /tmp/.mount_Slic3riOmLnh/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3riOmLnh/usr/lib/local-lib/lib/perl5/Slic3r/GUI.pm line 37.
Compilation failed in require at (eval 84) line 1.

me@host:~$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="16.04.1 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.1 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
UBUNTU_CODENAME=xenial
```

lordofhyphens, 2017-09-02:
Do you have freeglut3 installed on your local system? 

I'll add it to the list of packages for travis to download, though.

probonopd, 2017-09-02:
I don't think it comes with Xubuntu by default, so no, I don't have it on my system.

lordofhyphens, 2017-09-02:
There, it's added now. https://dl.slic3r.org/dev/linux/Slic3r-1.3.0-dev-8df65c1-x86_64.AppImage

probonopd, 2017-09-02:
Thank you very much. It is working for me now.

closed 2017-09-02 by probonopd

mderouss, 2018-11-29:
I'm trying to launch the 1.3.0 AppImage ( just downloaded from https://slic3r.org/releases/1.3.0/ ) on Xubuntu 18.04 with freeglut 3.0 installed. The GUI does not launch, and the following error is shown on the console -

libGL error: unable to load driver: nouveau_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: nouveau
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1734 error_code 2 request_code 155 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

After this error is produced, an AppImage process is still listed -

l 4567 ?        Ss     0:00 /opt/AppImage/Slic3r-1.3.0-x86_64.AppImage

The system is using nouveau drivers with an Nvidia quadro 4000 card. The Prusa edition Slic3r ( 1.41.2 ) AppImage seems to be working fine, as are other AppImages. 

Would you prefer me to raise a new issue here ?

lordofhyphens, 2018-11-29:
Yes, new issue please. Also please check with dl.slic3r.org/dev/linux

It probably has something to do with shipped libraries.

probonopd, 2018-12-30:
Please try:

```
Downloads/Slic3r-master-latest.AppImage --appimage-extract
wget https://raw.githubusercontent.com/AppImage/AppImages/master/functions.sh
. ./functions.sh 
cd squashfs-root/
delete_blacklisted 
./AppRun 
```

Does it run then?