slic3r/Slic3r #4043: Feature Request: sequential printing + spiral vase

JesusFreke, 2017-06-22:
### Version
v1.34.1-prusa36-linux64

### Operating system type + version
Ubuntu 14.04.5

### Behavior
1. Plate multiple objects 
2. Turn on "Spiral vase" option
3. Turn on "Complete individual objects" option
3. "Slice Now"
4. Get error saying "The Spiral Vase option can only be used when printing a single object."

It makes sense not to allow multiple objects when not using sequential printing, of course. But there shouldn't be any reason you can't spiral print one complete object and then spiral print the next complete object.

lordofhyphens, 2017-06-22:
Reminder: we're not prusa3d, so please write issues against dl.slic3r.org builds.

You're in luck this time as I am fairly certain that this behavior is the same ;)

JesusFreke, 2017-06-22:
Sorry about that! I just assumed this wasn't prusa-specific behavior and didn't even consider whether I should verify it on the "pure" slic3r build. I should have known better :)

lordofhyphens, 2017-06-22:
Huh, this should be already a thing. 

PrintConfig.cpp:679

JesusFreke, 2017-06-22:
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.cpp#L679 ? Hmm. doesn't look related to spiral/sequential slicing, but I'm not familiar with the codebase..

lordofhyphens, 2017-06-22:
That line is the check to do spiral vase + multiple complete objects. It's incorrectly implemented. Fixed now.

lordofhyphens, 2017-06-22:
Wait about 20 minutes or so and the AppImage and build will be ready at http://dl.slic3r.org/dev/linux (aa20fb6 will be present in the name)

lordofhyphens, 2017-06-22:
https://travis-ci.org/alexrj/Slic3r/builds/245967010 if you want to watch it go ;)

closed 2017-06-22 by lordofhyphens

JesusFreke, 2017-06-22:
Oh, you meant Print.cpp:679 :)

Anywhoo, Thanks! I'll give it a shot tonight.

lordofhyphens, 2017-06-22:
http://dl.slic3r.org/dev/linux/Slic3r-1.3.0-dev-837be65-x86_64.AppImage
http://dl.slic3r.org/dev/linux/slic3r-1.3.0-dev-837be65-linux-x64.tar.bz2

JesusFreke, 2017-07-03:
This is now working nicely with the fix. Thanks! :)

commit aa20fb65 (2017-06-22): Bugfix: refer to getBool() instead of getting a reference to the object itself for config options.
Fixes #4043