slic3r/Slic3r #4045: 3D Honeycomb infill -- update to switch between horizontal/vertical at squares

gringer, 2017-06-24:
[I hope it's okay for me to cross-post this to both Cura and Slic3r at the same time]

I've been working on an optimisation of the 3D Honeycomb code after experimenting with creating a large version of the honeycomb structure on my 3D printer (which hopefully will print quicker with fewer bugs). If the print alternates between horizontal and vertical at each square, it is able to print a fully-supported honeycomb structure with half as much filament and only sweeping through one direction. Here's the R code that I've used to test/debug my code for this updated method. It takes as input two variables: `el` - the length of the edge of the squares (which is the same as the bridging distance), and `zi` - the current height of the layer. It currently only prints 4 rows/columns, but it's easy enough to modify the _xofs_ `for` loop to sweep through the width of a print, incrementing by `2*el` each time:

```
#!/usr/bin/Rscript
library(rgl);

el <- 5; ## edge length
ch <- 4*el / sqrt(2); ## cycle height
pl <- el*4; ## path length

frame <- 0;
plot3d(NA, xlim=c(-10,40), ylim=c(-10,40), zlim=c(-10,40),
       box=FALSE);
for(zi in 0:80/3){
    ##for(zi in c(16,32) / 2){
    ## sawtooth wave function
    dl <- (abs(((zi*4) %%  (4*ch)) - (2*ch)) - ch) / 2; ## diagonal length
    dc <- floor(((zi*2) %% (2*ch)) / ch); ## diagonal cycle
    dox <- dl / sqrt(2); ## diagonal x offset
    doy <- abs(dox); ## diagonal y offset
    ol <- (pl - 2*doy)/2; ## orthogonal length
    for(xofs in c(0,el*2,el*4,el*6)){
        if(dc == 0){
            lines3d(x=rep(c(dox/2,dox/2,-dox/2,-dox/2,dox/2),2)+xofs,
                    y=rep(c(0, ol/2, ol/2+doy, ol*1.5+doy, ol*1.5+doy*2),2) +
                        rep(0:1*pl,each=5),
                    z=rep(zi,10), col=rainbow(40)[zi+1]);
        } else {
            lines3d(y=rep(c(dox/2,dox/2,-dox/2,-dox/2,dox/2),2)+xofs+el,
                    x=rep(c(0, ol/2, ol/2+doy, ol*1.5+doy, ol*1.5+doy*2),2) +
                        rep(0:1*pl,each=5)-el,
                    z=rep(zi,10), col=rainbow(40)[zi+1]);
        }
        dox <- dox * -1;
    }
    rgl.snapshot(sprintf("truncated_octahedron_%03d.png",frame));
    frame <- frame + 1;
}

```
Here's a simulated print animation to demonstrate the above code:

[image]

lordofhyphens, 2017-06-24:
Pull requests move quicker than feature requests ;)

gringer, 2017-06-25:
Yes, I'll do this up as a pull request eventually; just putting this up as a reminder for me, and in case anyone else wants to have a go at it.

lordofhyphens, 2018-01-06:
It looks pretty good, I'll merge the PR.

lordofhyphens, 2018-01-06:
Merged.

closed 2018-01-06 by lordofhyphens