slic3r/Slic3r #4047: Incorrect Temperature settings in GCODE generation

MajorDeth64, 2017-06-24:
### Version
_Version 1.3.0-dev 2017.06.22.1246.aa20fb6

### Operating system type + version
Windows 10 Pro

### Behavior
* _Describe the problem_
In Filament Settings | Filament | Temperature (C) I set the temperature for extruder to "X" for both input boxes. When GCODE is generated the temperature is X minus 5 degrees. This happens on all stls I slice.

Additionally it would also be nice if the temperature GCODE was added last and after any of the codes I enter in the Custom GCODES sections for all areas. That way I can run autolevelelling/bed compensation routines at my hot end Standby temperature, bed full temperature so I don't ooze dots all over my print bed and ooze doesn't interfere with levelling/compensation routines.

[config.zip](https://github.com/alexrj/Slic3r/files/1099959/config.zip)

lordofhyphens, 2017-06-24:
Can't reproduce.
This is with your config and a 20mm box on 13f121e3d952cca33b1ee5fdd73c12d85bf54d0e.
````
M107
M190 S65 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature
G1 Z5 F5000 ; lift nozzle

; Filament gcode

M109 S215 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; LAYER:0
````

On the second question, this is already a thing. See the tooltip for printer start gcode. 
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.cpp#L1286

closed 2017-06-24 by lordofhyphens

lordofhyphens, 2017-06-24:
In the future, please put separate things in separate issues; it helps us to not lose items when one is done and the other is undone.