slic3r/Slic3r #4053: Opening some stl files may crash crash Slic3r

EastonJ, 2017-07-02:
### Version
1.2.9-1120-g13f121e3

### Operating system type + version
Archlinux, freshly updated.

### Behavior
Slic3r crashes when it opens Juno.stl (https://nasa3d.arc.nasa.gov/detail/eoss-juno):  *** Error in `perl': realloc(): invalid next size: 0x00000000053f8710 ***

Other models seem fine.

#### STL/Config (.ZIP) where problem occurs
My config (I didn't change anything) and the offending .stl and  are included in files.zip. Terminal output is in crash.txt
[crash.txt](https://github.com/alexrj/Slic3r/files/1117506/crash.txt)
[files.zip](https://github.com/alexrj/Slic3r/files/1117511/files.zip)

atoms118, 2019-05-27:
I'm having the same issue with some STL files exported from Blender (not all though, I'm not sure what the difference between the ones that work and the ones that don't are) and also with this file: https://www.thingiverse.com/thing:3626710

I'm running Slic3r 1.3.0 on Gentoo.

dnandha, 2019-08-19:
Had the same issue, turned out my model was broken (needs to be one manifold). Fixed using [https://service.netfabb.com/service.php](https://service.netfabb.com/service.php).

atoms118, 2019-08-19:
This was my issue as well; that said, it would be much better to bring up an error message or at least log something sensible to stdio rather than crashing with a cryptic error message.

benjamin-garn, 2020-04-14:
One more example: https://www.thingiverse.com/thing:2319753 --> lcd12864top.stl
No problem to slice using cura, but Slic3r crashes "realloc(): invalid next size"