slic3r/Slic3r #4055: measurement tool

wjflier, 2017-07-05:
### Version
1.29

uld love to see a measurement tool or at least the grid numbered with more detail.

Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-07-05:
So what is your use case for this feature (aka why do you want it)?

wjflier, 2017-07-05:
best example is scaling a model to a precise size.  For now I import to sketcher to measure.  Maybe I'm missing something but counting squares in the grid doesn't work well.

lordofhyphens, 2017-07-05:
You can already scale to a specific size in the UI though, and iirc the
tiles are 10 units (commonly mm).

wjflier, 2017-07-05:
I'm aware of that, but scaling to a precise measurement is not possible, it's only a guess.  If I need to scale to 13mm or 43mm it's not very accurate to do.  I appreciate that it might not be easy to add more precise measurements, I was only asking.

lordofhyphens, 2017-07-05:
Your original feature request was scant on useful information other than "I
would like to be able to measure things".

Since that capability isn't obviously useful to me (because I am not you),
more information is needed in as to how it would be used (and why it would
be useful). What problems are you trying to solve?

The link at the bottom of the original post is a good general guide to
writing feature requests, I suggest you take a look at it and revise your
request accordingly.

wjflier, 2017-07-06:
are you serious?

lordofhyphens, 2017-07-06:
In 1.3.0-dev (https://dl.slic3r.org/dev/win) you can right-click and "Scale to Size" in all three axes that let you scale to at least the nearest floating point number (subject to fp rounding error). It certainly is more accurate than your printer.

wjflier, 2017-07-06:
I see, that's a great improvement, nice job. and Thanks
Bill

wjflier, 2017-07-06:
One last question.  Is there anyway to lift the model on the z axis?  With multi-color printing becoming more popular, I think this would be a great add.

lordofhyphens, 2017-07-06:
I don't know of one, do a quick search on the issue tracker and post a request  (with the problem you want to solve and use cases, etc).

closed 2017-07-06 by lordofhyphens

bradosav, 2019-11-05:
Hi there! I realize this is a closed issue, however there is merit in further discussion. Slic3r is a wonderful software, and the scaling option to defined outside dimensions is a great feature. Sometimes it is necessary to measure a different dimension. I am trying to print some hexagonal modules for a spice rack (i.e. https://www.thingiverse.com/thing:1743145). It is important that my spice bottles fit inside the modules. I've tried many approaches, printed many trials, but what would be nice is a tool to measure other dimensions, in my case the inside dimensions between two parallel faces of the hexagonal module. Thanks in advance for any thoughts!

supermerill, 2019-11-05:
Quick hack: import the model of your rack (two simple box spaced by the good amount of mm), and use it to check.

foreachthing, 2019-11-07:
@bradosav : Use [Blender](https://blender.org) for [such things](https://www.youtube.com/watch?v=OZECjQS5yVo). And, if needed, you can scale your model right in Blender - any way you need it! There's a handy "3D Print" addin that exports your model as stl with one click as well. Also, the Blender-Addin can check your model (manifolds, intersections, etc.) and even might be able to fix your model.