slic3r/Slic3r #4056: missed line number

cdedwards, 2017-07-09:
1.3-dev
Windows 10 latest
`
[image]
`
while on a 4 hour print, about half way through, I received an error saying the last line number was not the last line. This caused the print to stop. I was able to restart the print, by entering M110 108572. I am pretty sure this does not continue exactly where it left off, as the printer scooted across the bed before continuing.

kromar, 2017-08-03:
since yesterday i cant even start a print anymore due to this error, as soon as im connected to the printer with slic3r then the console gets spammed with this message and when the print is supposed to start my windows system throws a blue screen (driver irq or something like that).
other softwares do not have any issue printing the exported gcode or the same model. can someone please tell me why this suddenly happens and how this can be fixed? thanks

lordofhyphens, 2017-08-03:
Send m110 s0 and your printer should be fine, or reset your printer.

lordofhyphens, 2017-08-03:
The bluescreen is definitely odd and probably has to do with a weird system
state

lordofhyphens, 2017-08-03:
@kromar @cdedwards could you also mention the printer/firmware you're using? I suspect it will become relevant.

lordofhyphens, 2017-08-04:
So I set up an arduino board with Marlin 1.1 release and looked into how the host sends gcode and determines the current line number.

The error you're seeing is because Marlin expected a different line number than what was sent by Slic3r. How that is ending up, I'm not sure.

I suspect the serial port is getting into a weird state.

lordofhyphens, 2017-08-04:
The missed line number itself is because (probably) your fw dropped the ball. I don't know if we're correctly resending. 

@kromar can you see if #4079 helps at all?

lordofhyphens, 2017-08-04:
@kromar @cdedwards if you can build from source, add -DDEBUG_SERIAL to the array line 22 of xs/Build.PL and rebuild. That will give you better debugging output from Slic3r.

kromar, 2017-08-04:
thanks for the information, i will see what i can do this weekend. as for the printer, its a wanhao i3 plus and i tried firmwares with different marlin versions 1.0./0 1.1.1 and both have the same issue as of a day ago, before i never got this.
what do you mean with resetting my printer? is there a special reset that im not aware of or just a factory reset?
and since i have no ide what the m110 s0 does i need to look that up or if you feel like it you could give me a short explanation
if i can provide any more information let me know.

lordofhyphens, 2017-08-04:
M110 N0 tells the printer to expect the next line # to be 0.

I am considering putting a check for a resend loop in that forces this to
be set if this condition occurs.

kromar, 2017-08-05:
#4079 does not seem to make a difference, i also get the error in the log. but the M110 N0 does remove it. i also noticed that i get a "Cannot resend 2 (oldest we have is 0)" in the console so maybe that gives any more clue.
im testing for the bluescreen now to see what exactly is happening at that point

kromar, 2017-08-05:
here is also the snippet from the log when the first line error happens, maybe you can see something out of the ordinary?

`echo:Marlin 1.0.0
echo: Last Updated: Jun  7 2017 12:59:05 | Author: (none, default config)
Compiled: Jun  7 2017
echo: Free Memory: 3679  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X81.00 Y81.00 Z400.50 E94.30
echo:Maximum feedrates (mm/s):
echo:  M203 X450.00 Y450.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E3000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S800.00 T800.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.40 E1.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P33.41 I1.47 D189.27
Error:Line Number is not Last Line Number+1, Last Line: 0
echo:SD card ok`

kromar, 2017-08-05:
this is the bluescreen that i get when the print is starting with the master and the log is spammed with the line error:

080517-11453-01.dmp	05-Aug-17 11:59:23	DRIVER_IRQL_NOT_LESS_OR_EQUAL	0x000000d1	ffffd700`00000003	00000000`00000002	00000000`00000001	fffff800`af74416c	CH341S64.SYS	CH341S64.SYS+416c					x64	ntoskrnl.exe+16c4c0					C:\WINDOWS\Minidump\080517-11453-01.dmp	8	15	15063	174,697

kromar, 2017-08-05:
i updated the usb serial driver and that seems to have fixed the bluescreen issue (fingers crossed) and with the #4079 it works for me so far, thanks a lot for that fix
[image]

Nuttavoot, 2018-02-14:
This is still open, I still have this line number error with the latest 1.3 release.

bubnikv, 2018-02-14:
slic3r just sends the initial temp inquire g codes before the marlin
firmware boots.

Nuttavoot, 2018-02-14:
Any way to fix it, now I can't print using USB.

bubnikv, 2018-02-14:
Likely a reasonable implementation would be to make the starting sequence firmware specific, and for some firmware types to wait a bit after the connection for the firmware to boot up.

lordofhyphens, 2018-02-14:
@Nuttavoot disconnect Slic3r host and power cycle the printer. 

Alternatively send the gcode to reset the line number  (see reprap wiki).

Nuttavoot, 2018-02-14:
@lordofhyphens tried both power cycle and gcode "M110 N0", all did not work.

Jedrusia, 2019-08-23:
I have a identical unsolved problem with my Ender-3 Pro and fw 1.1.6.
I can't print any models, only i can send full defined own start g-code (two test lines on left side my bed).
Who was resolved this problem ?