slic3r/Slic3r #4057: Multiple Extrusers with different diameters/extrusion multiplier

tps42, 2017-07-11:
### Version
          ALL
          example: 1.3.0-dev

### Operating system type + version
           windows

### Behavior
            Using different extruders with different filament diameter results in "random" results.
Example: 
extruder 0: diameter 3mm, extrusion multiplier 1.1
extruder 1: diameter 1.75mm, extrusion multiplier 1.2

Insert to "Printer Settings"/"Custom G-Code":
; filamentDiameter=[filament_diameter]
; filamentDiameter0=[filament_diameter_0]
; filamentDiameter1=[filament_diameter_1]
; filamentDiameter2=[filament_diameter_2] 
; filamentDiameter3=[filament_diameter_3]
; extrusionMultiplier=[extrusion_multiplier]
; extrusionMultiplier0=[extrusion_multiplier_0]
; extrusionMultiplier1=[extrusion_multiplier_1]
; extrusionMultiplier2=[extrusion_multiplier_2]
; extrusionMultiplier3=[extrusion_multiplier_3]

Slice 2 times: 
1. select first extruder in Positioning 
2. select second extruder in Positioning

Results are ... errrrr ... unexpected:
1. 
; filamentDiameter=3
; filamentDiameter0=3
; filamentDiameter1=7
; filamentDiameter2=1.75 
; filamentDiameter3=7
; extrusionMultiplier=1.1
; extrusionMultiplier0=1.1
; extrusionMultiplier1=1.2
; extrusionMultiplier2=1.1
; extrusionMultiplier3=1.1

i got multipliers korrekt but diameters wrong! "7"???? what the hell?

2. 
dont remember but it was not better than the first slice.

_Is this a new feature request?_
                no feature request

lordofhyphens, 2017-07-11:
Please include an exported configuration (CTRL+E) file.

tps42, 2017-07-16:
While exporting slic3r configuration i was able to investigate further:
This is not a slic3r bug. The problem emerges in combination with repetierhost (v2.01) only!

So please mark this one as invalid - sorry for that!

cu. tps42

closed 2017-07-16 by tps42

lordofhyphens, 2017-07-16:
As a reminder you can control your printer directly from Slic3r now. No need for RepHost ;)

tps42, 2017-07-18:
My printer is driven by repetierserver (network).Slic3r has only restricted support for network control (manual control of movement, temperatures, etc.) so using slic3r only is no option for me.
Thanx anyway!
Cu.tim