slic3r/Slic3r #4058: Free to wrong pool

marclefevre, 2017-07-12:
### Version
1.29

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 7 SP1

### Behavior
Free to wrong pool error message
Load the STL file and attempt to rotate it 45 degrees clockwise

* rotation
*crash
I got a screen grab.  It's attached

[crash_docs.zip](https://github.com/alexrj/Slic3r/files/1143115/crash_docs.zip)
[crash_screen_grab.pdf](https://github.com/alexrj/Slic3r/files/1143116/crash_screen_grab.pdf)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-07-12:
Please try to reproduce with build from dl.slic3r.org/dev/win

marclefevre, 2017-07-13:
I can't make this build fail the way that the previous one did.  Should I
continue to use the development version for now?

lordofhyphens, 2017-07-13:
Yes

marclefevre, 2017-07-16:
I've seen a lot of bugs in this dev version.  Do you want to hear about
them?  They include:

1) Can't move objects on the build plate
2) Once you turn off support, it won't turn back on again.  That is, you
can check the box but the support is not generated. I verified that the
original Slic3r I was using generates the expected support.
3) The "Generate support" check box in the right-hand pane in the preview
tab does not match the check box in the print Settings->Support material
dialog.
4) Probably not a bug but when you alternate between version 1.2.9 and this
dev version, the dev version adds double quotes around the path/filename to
a post-processing script.  This results in a bizarre windows script error
that says that the script is not executable.

Marc

lordofhyphens, 2017-07-16:
Separate issues go in separate posts.