slic3r/Slic3r #4059: Can't control or send file to printer, no live status feedback (but I can connect and see the printer-startup dump)

VanessaE, 2017-07-13:
### Version

1.3.0-dev, commit 13f121e3d952cca33b1ee5fdd73c12d85bf54d0e

### Operating system type + version

Debian 9 "Stretch"

### Behavior

The "Test" function under Printer Settings tells me the connection is okay, and I can indeed connect to the printer inside the Controller tab, after which I am shown the printer's usual status/info dump.

However, none of the Manual Control dialog's functions actually work - there is no reaction of any kind at all from the printer for anything I try.

If I try to jog without homing first, Slic3r will warn me, but trying to home doesn't actually do anything.  If I use the "Send manual command" function, it echoes the command I sent, to the log, but again produces no reaction from the printer.

Attempting to print a sliced object adds the print job to the queue, but clicking that queued job's "print" button does nothing.

If I close the Manual control dialog and disconnect from and then re-connect to the printer, I am again shown the printer's status dump.  I find this odd, because that doesn't happen in Pronterface - it only shows me the status dump if I reset the printer and then disconnect/reconnect to it.  Is Slic3r resetting the printer on connect?

Thing is, this worked recently - at least, the manual jog-by-dragging-the-mouse feature did.  I last tried that particular feature (which is really neat, by the way) some time before July 2.  I can't tell for sure when it broke; I've tried a few other commits, going back as far as 67d42f9e4f41744991c25c9707da317712f1f3f2, with no success.

Sometimes, I get live hotend/bed temperature readings to show up, but that's pretty rare, but manual control does not work in any case.

I think Slic3r is failing to keep the connection "alive"/"open", because if I connect with it, and then connect with Pronterface (while Slic3r is still "connected"), neither program shows any errors and I can even manually control the printer with Pronterface at that point.

If I try it the opposite way, that is connecting with Pronterface first, then connecting with Slic3r, I get the sort of behavior I would expect: Pronterface throws a couple of bad line number errors in its log, and Slic3r reports a connection failure.

#### STL/Config (.ZIP) where problem occurs

Doesn't matter if a model is loaded or not.
Config: [config.ini.zip](https://github.com/alexrj/Slic3r/files/1144954/config.ini.zip)

lordofhyphens, 2017-07-13:
Can you send m110 s0 with manual gcode in slic3r?

VanessaE, 2017-07-13:
It echoes ">> m110 s0" into the log, and otherwise does nothing.

VanessaE, 2017-07-14:
As @lordofhyphens suggested, I enabled serial debug and rebuilt.

I connected to the printer, saw the status dump in the log on the right.  Tried to home-X, didn't respond.  Tried to send M110 S0 followed by G28, no response.  Disconnected, closed Slic3r.

Here is serial.txt in its entirety:

```
>> N0 M105*39
<< start
<< echo:Marlin bugfix-1.1.x
<< echo: Last Updated: 2017-05-04 12:00 | Author: (none, default config)
<< Compiled: Jul 12 2017
<< echo: Free Memory: 5249  PlannerBufferBytes: 1312
<< echo:Hardcoded Default Settings Loaded
<< echo:  G21 ; Units in mm
<< echo:Filament settings: Disabled
<< echo:  M200 D3.00
<< echo:  M200 D0
<< echo:Steps per unit:
<< echo:  M92 X80.00 Y80.00 Z4000.00 E760.00
<< echo:Maximum feedrates (units/s):
<< echo:  M203 X450.00 Y450.00 Z2.50 E52.00
<< echo:Maximum Acceleration (units/s2):
<< echo:  M201 X5000 Y5000 Z100 E10000
<< echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
<< echo:  M204 P2000.00 R2000.00 T2000.00
<< echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
<< echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
<< echo:Home offset:
<< echo:  M206 X0.00 Y0.00 Z0.00
<< echo:Auto Bed Leveling:
<< echo:  M420 S0
<< echo:PID settings:
<< echo:  M301 P14.86 I1.09 D50.77
<< echo:  M304 P207.96 I29.38 D367.98
<< echo:Z-Probe Offset (mm):
<< echo:  M851 Z-0.25
<< echo:Linear Advance:
<< echo:  M900 K20.00 R0.00
DISCONNECTED
```

It seems to be as I suspected - Slic3r disconnects from the printer after receiving the status dump.  As you can see, my home-X, M110, and G28 commands do not appear in the log.

Pronterface log shows almost exactly the same text after a printer reset, except there is no "N0 M105*39" in the log.

VanessaE, 2017-07-14:
A bit of googling says the N0 M105*39 is just raw gcode for line 0, M105, checksum value of 39, but why's it being sent before connecting?

VanessaE, 2017-07-14:
For comparison, here's Pronterface's logfile with "debug" enabled:

```
tail: '/home/vanessa/pronterface-log.txt' has appeared;  following new file
2017-07-14 00:28:18,826 - Connecting...
2017-07-14 00:28:18,878 - SENT: M105
2017-07-14 00:28:19,748 - RECV: start
2017-07-14 00:28:19,749 - Printer is now online.
2017-07-14 00:28:19,756 - RECV: echo:Marlin bugfix-1.1.x
2017-07-14 00:28:19,757 - RECV: echo: Last Updated: 2017-05-04 12:00 | Author: (none, default config)
2017-07-14 00:28:19,760 - RECV: Compiled: Jul 12 2017
2017-07-14 00:28:19,765 - RECV: echo: Free Memory: 5249  PlannerBufferBytes: 1312
2017-07-14 00:28:19,769 - RECV: echo:Hardcoded Default Settings Loaded
2017-07-14 00:28:19,770 - RECV: echo:  G21 ; Units in mm
2017-07-14 00:28:19,773 - RECV: echo:Filament settings: Disabled
2017-07-14 00:28:19,777 - RECV: echo:  M200 D3.00
2017-07-14 00:28:19,778 - RECV: echo:  M200 D0
2017-07-14 00:28:19,778 - RECV: echo:Steps per unit:
2017-07-14 00:28:19,781 - RECV: echo:  M92 X80.00 Y80.00 Z4000.00 E760.00
2017-07-14 00:28:19,785 - RECV: echo:Maximum feedrates (units/s):
2017-07-14 00:28:19,789 - RECV: echo:  M203 X450.00 Y450.00 Z2.50 E52.00
2017-07-14 00:28:19,790 - RECV: echo:Maximum Acceleration (units/s2):
2017-07-14 00:28:19,793 - RECV: echo:  M201 X5000 Y5000 Z100 E10000
2017-07-14 00:28:19,801 - RECV: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
2017-07-14 00:28:19,806 - RECV: echo:  M204 P2000.00 R2000.00 T2000.00
2017-07-14 00:28:19,815 - RECV: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
2017-07-14 00:28:19,819 - RECV: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
2017-07-14 00:28:19,822 - RECV: echo:Home offset:
2017-07-14 00:28:19,823 - RECV: echo:  M206 X0.00 Y0.00 Z0.00
2017-07-14 00:28:19,826 - RECV: echo:Auto Bed Leveling:
2017-07-14 00:28:19,828 - RECV: echo:  M420 S0
2017-07-14 00:28:19,831 - RECV: echo:PID settings:
2017-07-14 00:28:19,831 - RECV: echo:  M301 P14.86 I1.09 D50.77
2017-07-14 00:28:19,834 - RECV: echo:  M304 P207.96 I29.38 D367.98
2017-07-14 00:28:19,835 - RECV: echo:Z-Probe Offset (mm):
2017-07-14 00:28:19,838 - RECV: echo:  M851 Z-0.25
2017-07-14 00:28:19,839 - RECV: echo:Linear Advance:
2017-07-14 00:28:19,850 - RECV: echo:  M900 K20.00 R0.00
2017-07-14 00:28:21,884 - SENT: M105
2017-07-14 00:28:21,891 - RECV: ok T:25 /0 B:25 /0 @:0 B@:0
2017-07-14 00:28:24,900 - SENT: M105
2017-07-14 00:28:24,905 - RECV: ok T:25 /0 B:25 /0 @:0 B@:0
2017-07-14 00:28:27,904 - SENT: M105
2017-07-14 00:28:27,912 - RECV: ok T:25 /0 B:25 /0 @:0 B@:0
2017-07-14 00:28:30,197 - Disconnected.
```

To get a clean capture, I started Pronterface, told it to reset the printer, then closed it, cleared the log file, started Pronterface again, enabled "debug communications", connected to the printer, and let it sit for a few seconds before I disconnected again.

VanessaE, 2018-12-04:
Not sure what changed, but this feature works now (commit 21eb603cc16946b14e77d3c10cbee2f1163503c6).

closed 2018-12-04 by VanessaE