slic3r/Slic3r #4060: Layer height for multiple extruder

Pydo44, 2017-07-14:
Hello
i new some new feature

i use two extruder one with PLA and an other PVA
but i need to put the height for the PVA a little bit more than the PLA
Like that the PLA can grab easily the PVA on a plane support

thanks for your job

Jasper2991, 2017-07-20:
you can add a modifier for diffrent STL's 
This overwrites the settings in print settings for the selected STL. 

Alternatively when your PVA part is Slic3r generated support, set the print settings layer height to PVA layer height and overwrite the layer height for your PLA part by adding the modifier with PLA layer heights.

I have not tested this recently but it should work.

alexrj, 2017-07-21:
@Pydo44, before asking for new features you should describe the problem you have. Problem first, solution after. Nobody is doing volunteer work for goals which are not clear.

So far, I can only guess things.
I guess you are using PVA for support material and PLA for the object. Support material layers are already generated thicker, regardless of the configured layer height which is used only for the object.

closed 2017-07-21 by alexrj