slic3r/Slic3r #4061: Porting to c++ is slow and cumbersome

fivethreeo, 2017-07-18:
Been looking at porting all of slic3r to c++.

But making xs modules for new c++ code seems silly since perl is going away and the building of the modules takes foreversince all has to be rebuilt for every change.

Should have a branch with a worklow for doing gui and tests in c++.

For example using cmake and CATCH.

lordofhyphens, 2017-07-18:
You mean xsgui and the src dir?

As for tests we are keeping them in perl for now. Helps to make sure the
interface is clear.

lordofhyphens, 2017-07-18:
The next big need for porting is support, by the way.

Also, you don't need to rebuild everything; run Build in the xs dir.

closed 2017-07-18 by lordofhyphens

fivethreeo, 2017-07-18:
Not to argue, but build in xs dir builds all the xs modules, right? So to change test/change test it takes 2 min. Keeping the interface clear? I dont get what you mean by that. :) But if I started doing the porting does  the suggested tools seem sensible so I dont start doing work that wont be used.

lordofhyphens, 2017-07-18:
It only builds things that have changed. Install the module to test it, and you can run the tests from there too.

The src dir is a separate path that uses cmake.

It would probably behoove you to look at the other porting-related issues as well (I have been working on xsgui items, specifically the options config stuff)

I will let @alexrj jump in ;)

lordofhyphens, 2017-07-18:
Workflow recommendation: Pick one piece, make a pull request. Wait for it to get reviewed and accepted before you do something else that depends on it. 

It will be much easier to get your contributions in if you do this.

bubnikv, 2017-07-18:
> The next big need for porting is support, by the way.

Supports are reimplemented into C++ / revamped in the Prusa fork for a long
time. They are already stable, very fast and fully parallelized.

You can rebuild the XS module by running
perl xs/Build install

but it is not always reliable. Namely, it only correctly handles the
changes in the C/C++ modules, not in the headers. But often it works. So I
do not run the complete build process that often.

Also @CReimer <https://github.com/CReimer> works on rewriting the XS
compilation into cmake.
https://github.com/prusa3d/Slic3r/issues/420
https://github.com/CReimer/slic3r/tree/makefile

platsch, 2017-10-20:
@lordofhyphens @alexrj just took a closer look at the prusa3d cmake implementation. Seems not to be too much effort to backport, do you have any important objections if someone would take the effort or do you already plan to do so?
I'm still unsure how good it would / should integrate with the src cmake.
the perl build-system is no fun at all...

lordofhyphens, 2017-10-20:
I would look at a pull request and merge it in if it doesn't completely
break the build services.

lordofhyphens, 2017-10-20:
I don't think it should integrated with the src cmake at all. Eventually
the src cmake can be integrated with it or w/e.

If someone wants to pick a test framework (CATCH probably is not a bad
choice) and port the XS tests to it at the same time I'd also likely merge
it.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

platsch, 2017-10-20:
test framework would have been my next question. It just doesn't make sense to write an xs interface solely for testing. My first suggestion would be the google test framework, but I have no experience with it and not investigated any further

lordofhyphens, 2017-10-21:
@platsch while I agree with you, @alexrj prefers to have the xs interface to force people to think about the interfaces. 

https://github.com/philsquared/Catchhttps://github.com/philsquared/Catch for catch. It's header-only, which is handy and I think plays nice with CMake.

lordofhyphens, 2017-10-24:
@fivethreeo @platsch #4168 as a FYI. I'm still working on getting the build server to function properly.

lordofhyphens, 2017-10-24:
I'd lean towards CATCH myself as a C++ test framework, would prefer the test interfaces be the public interfaces be the same as the public interfaces of libslic3r for most things though.

platsch, 2017-10-24:
Oh yes! Thats good news. Was just about to start with this myself, you saved me a lot of work! Great.

platsch, 2017-11-03:
Just forked your cmake branch to add experimental CATCH support: https://github.com/platsch/Slic3r/tree/catch-testing
works well in my setup, no build server integration yet. `make test` invokes CATCH and perl tests.

I ported the xs/t/03_point.t to have a real test case, the perl test is still in place.
Also added a `<<` operator to Point to get meaningful expansions in failed tests, we should probably do this for most classes.

Currently only xs tests are covered by the cmake integration. What exactly is the difference between the tests in t vs xs/t? I don't see a clear separation here...

Comments are very welcome, don't want to port more tests unless we are actually going to use them.

fivethreeo, 2017-11-04:
Nice work. About what I wanted to see in terms of being able to not have to learn internals of perl modules  and bindings. About the interfaces I get the need for think about interfaces, but as I understand it xs is for bindings to perl and requires Build.pl that is hard to grok. So this gets my vote. But up to the maintainers to decide :)

lordofhyphens, 2018-06-22:
@platsch @fivethreeo just as FYI, I am using CATCH for cppgui and it will be the framework used.