slic3r/Slic3r #4062: Something  wrong with gcode export in files

Tinchus2009, 2017-07-18:
### Version
_1.2.9-1120-g13f121e

### Operating system type + version
win10

### Behavior
Im useing octoprint on a raspberry. Useing samba, I map a network folder , a watched folder. So, in theory, I just need to export a gcode file in that network folder (the folder is located in the raspberry), and octoprint automatically sees the file. Well, this is not working with slicer. I meanL slic3r exports the gcode to that folder and nothing happens. I though it was a problem with my setup, but useing the same file, if I export the gcode from CURA and simplify3d, it works. So the problem probabl is not the setup but something related in the export with slic3r, this is my conclusion since the export useing the same file, same setting from other slicers works ok.

lordofhyphens, 2017-07-18:
Is it just network folders?

Also... we have octoprint integration already.

Tinchus2009, 2017-07-18:
May be I was not clear, sorry: slic3r exports the file, the .gcode file to the network folder. The file is written there. I can see it, I can see it too if I log in into the raspberry. But octoprint cant see the file. You can say: ok, it is a problem in the setup or it is a problem of octoprint... but the same stl, sliced with cura and exported, is perfectly seen. The same happens with simplify3d. With other slicers, it works. 
What do you mean with octoprint integration? I just did a git pull and Im updated to latest dev version but cn see anythong related to octorpint

lordofhyphens, 2017-07-18:
There's information for an API key in Slic3r's Printer config (has been for quite some time now) and a server IP. Fill it in and you will get a "Send to Printer" button on the main UI.

Re: Your stated problem, is it still an issue if you export it locally and copy it manually? I'm suspecting that there might be an OS-level lock getting made on the file and either it's not being released properly (which is very likely out of the control of Slic3r, but that is able to be checked via inspection of GcodeWriter) or something weirder is happening.

lordofhyphens, 2017-07-18:
One other thing I'd suggest (just to factor out any weirdness in your build environment) is to grab a build off of http://dl.slic3r.org/dev/win, they are ordered newest first.

Tinchus2009, 2017-07-18:
If I export the gcode from slic3r, them from the file manager I move/copy the file to the network folder, then all works, so something probably is not being released by sli3r at the end of the export? If the problem would be from the OS side, I would have the same problem with the other slicers I think, but this happens only with slic3r. I will try getting the latest of your builds and check again

Tinchus2009, 2017-07-18:
Nope... with the latest build still doesnt work. Also, after exporting, I closed slic3r. That should release any lock comeing from the OS? because after doing that still octoprint seems not to see the file.

jluis, 2017-07-18:
I've seen, in my workplace, frequent random network looks, from windows workstations, that persist until the workstation it is rebooted  but I've never able to generate a case to report the bug to Microsoft

Tinchus2009, 2017-07-18:
no, rebooting doesnt help. Anyway, the octoprint integration works perfect and gives me the same result, so I will use that. But definitly I think it is somenthing really weird and it is specific to slic3r because I have just done a funny test: useing notepad, I created a .text file, and saved it in the watched folder... octoprint sees that file... lol
The problem is there but there is a workaround already in place so you can close the post if you want

lordofhyphens, 2017-07-18:
It is definitely weird and worth looking into.

jeder73, 2017-08-17:
I think the problem could only be the file metadata. Perhaps file creation date in the future or any attribute (hidden, system) is translated incorrect by samba or windows SMB client or such a thing...because any probs with the file content would not have been "repaired" by copy (Byte order mark or so..), copy only copies Last Changed but creates a new file...
btw: If you talk about "file manager" do you mean Explorer on W10 or some third party file manager? Perhaps there is any unknown "intelligence" in that copy process? (BOM/ANSI/Windows1252/line break conversion or so...)