slic3r/Slic3r #4064: advanced parameters in Slic3r :  exrusion width

Insafa, 2017-07-18:
Hello everyone,
I'm a beginner in the 3D printing, I hope you help me solving the problem I have.
Well, I have a **prusa i3** in the laboratory, and I have to do some research, namely parameter optmisation. 
The problem is the parameters in the "exrusion widthxrusison width**" tab, I don't know what do they mean and how should I choose them. for example the "perimeter",  is it the same as "contour width" ? and which values it can take? 
[image]

Jasper2991, 2017-07-20:
Extrusion with is the calculated line with.
to optimize a print this with can be set for different part of a print:

First layer: value used for the first layer of the print.
Perimeters: the value used for contours as specified in your drawing as "contour with".
External perimeters: in my understanding this is only the most outer contour of the print. 

I think you get the point.

take a look at http://manual.slic3r.org/ and https://github.com/alexrj/Slic3r/wiki/Documentation 
there is a lot of documentation to help you along.

lordofhyphens, 2017-07-20:
And look at PerimeterGenerator.cpp implementation.

alexrj, 2017-07-21:
@Insafa, you mentioned a problem but you didn't describe it, so we can't really help.
I recommend to leave Perimeter Extrusion Width to the default value.

closed 2017-07-21 by alexrj