slic3r/Slic3r #4065: Deleting post procesing scripts crashes slic3r

Tinchus2009, 2017-07-19:
### Version
1.2.9-1120-g13f121e

### Operating system type + version
win 10 64b

### Behavior
In the post processing scripsts section, once you added a scrip, you cant delet it anymore without crashing slic3r, so in fact, you never cant delet it from that printing profile.

superjamie, 2017-11-23:
Confirming this still happens in `Slic3r-1.3.0-dev-8ca6171-x86_64.AppImage`

superjamie, 2017-11-23:
You can work around it by manually editing the ini file and changing

~~~
post_process = " " 
~~~

to

~~~
post_process =
~~~

bonm014, 2017-12-28:
Where can I find that ini file?

lordofhyphens, 2017-12-28:
On Linux / Mac, ~/.Slic3r

For Windows it is buried in your AppData folder structure. Searching for
"Slic3r" should find it.

lordofhyphens, 2017-12-28:
Another workaround is to hit Ctrl-e for a config export, edit that and then load it and save over the old Print profile.

bubnikv, 2017-12-28:
- Windows: "C:\Users\your_user_name\AppData\Roaming\Slic3r" or
   "C:\Documents and Settings\your_user_name\Application Data\Slic3r"
   - Mac: "~/Library/Application Support/Slic3r/"
   - Unix / Linux: "~/.Slic3r/"

lordofhyphens, 2017-12-28:
@bubnikv @alexrj can you think of a reason offhand of why we wouldn't want to trim excess whitespace from ConfigOptionString objects?

I can't think offhand of a use case where you would want to keep excess whitespace before and after.

lordofhyphens, 2017-12-28:
The other solution is to check for the problem cases in _update.