slic3r/Slic3r #4066: Custom layer change G-code before mid-layer tool-change

Jasper2991, 2017-07-20:
### Version
1.2.9

### Operating system type + version
Windows 10 
Version 1607 (OS Build 14393.1480)

### Behavior
Slic3r outputs (custom) layer change G-code before tool change mid layer. 

* _Steps needed to reproduce the problem_
two stl files (included in attached ZIP)
One set to extruder 2 with modifier:
[image]

Set Aditional custom g-code in printer settings:
[image]

* _Expected Results_
Tool change mid layer without aditional g-code, just G92, and Tn command.
At effective layer change (Z change/next slice):
Custom layer change G-code Before and after actual layer change and tool change comand when needed.

* _Actual Results_
Custom layer change commands and G1 Zn command to current Z height, before mid layer tool change commands. 
this part of the Gcode is the tool change during printing of layer 1 that started at Z0.050:
"
G1 X24.089 Y67.589 E5.95361
G1 X29.836 Y67.589 E6.22205
G4 P300 ; before layer change, layer: 1
G1 E6.14351 F120.00000
G92 E0
G1 Z0.050 F600.000
G4 P100 ; After layer change, layer: 1
G1 E-0.07854 F120.00000
G92 E0
G4 P200 ; tool change
T0
G92 E0
G1 X31.021 Y63.582 F600.000
G1 E0.15708 F120.00000
G1 X25.737 Y58.298 E0.50611 F600.000
G1 X25.186 Y58.298 E0.53185
"
A layer change is processed where i would expect only a tool change. 

This results in custom layer change G-code executed twice per effective layer for a dual-extruder setup. 

#### STL/Config (.ZIP) where problem occurs
[layer change G-code after tool-change.zip](https://github.com/alexrj/Slic3r/files/1162176/layer.change.G-code.after.tool-change.zip)

lordofhyphens, 2017-07-20:
Just to be sure, can you confirm this on 1.3.0-dev?

Jasper2991, 2017-07-20:
Thank you for the fast response. 

I can Confirm this happens in 1.3.0-dev. 
(Zip download Slic3r-master.2017.06.24.1248.13f121e.64bit)

There is a diffrence in layer count.
In the 1.3.0-dev generated G-code the Layer numbers change on tool change mid layer. 

So in layer 2 (Z0.1) this happens:
Start of layer 2
G1 X29.411 Y46.836 E2.01878
G4 P300 ; before layer change, layer: 2
G1 E1.94024 F120.00000
G92 E0
**G1 Z0.100 F600.000** 
G4 P100 ; After layer change, layer: 2
G1 E-0.07854 F120.00000
G92 E0
G4 P200 ; tool change
T1
G92 E0
G1 X39.031 Y47.926 F600.000

followed by a tool change that increases the layer_num placeholder. 

G1 X29.836 Y67.589 E6.00018
G4 P300 ; before layer change, layer: 3
G1 E5.92164 F120.00000
G92 E0
**G1 Z0.100 F600.000**
G4 P100 ; After layer change, layer: 3
G1 E-0.07854 F120.00000
G92 E0
G4 P200 ; tool change
T0
G92 E0
G1 X31.114 Y66.077 F600.000

Is it possible that by having the modifier active, in STL dependent settings, to change the used extruder for a STL: the layer counter will act as if the modifier for layer height of a part is active?

lordofhyphens, 2017-07-22:
I can't quite parse what you said there, is it possible to explain it with
pictures?

Jasper2991, 2017-07-24:
I tried to make a clear overview of the problem
[image]

The produced G-code contains twice as much custom layer change G-code as expected.
You can verify this in the G-code attached to the first post.

lordofhyphens, 2017-07-24:
Thank you, that makes it much clearer.

Jasper2991, 2017-07-25:
I have found that this also happens when 2 stl's are loaded on buildplatform and no multimaterial settings are used.

The 'extra' set of custom layer change G-code is then written halfway in the regular 'G1 X Y E' code.
this seems to be the exact point the extruder switches to the next SLT/body/part.

Jasper2991, 2017-08-04:
I tested some things in 1.1.7 as this is the last release with muti-material AMF support AND object editor window.

I generated a AMF file with two STL's and then sliced it (no modifiers) . it resulted in a multi-material Gcode with correct layer counting and layer change Gcode

When loading the platter with the two stl's (not the AMF) and adding a default extruder modifier, in the object editor window, to the object (not part) of one STL's , the Gcode contained unwanted layer change Gcode. 

When loading the same STL's and slicing them without any modifiers (single material/extruder) the Gcode contained unwanted layer change Gcode. 

My guess is that this issue has to do with using two objects/stl's because when loading one AMF (object) with two parts the problem dos not occur. 

i would like to see this issue solved as i want to synchronize my printer with a secondary proces by adding a M42 and M226 command on every layer change. With this 'bug' of unwanted custom layer change Gcode this is near impossible to do without editing every layer change in the Gcode manually.

stepping back to 1.1.7 is not the best option because i use several features that are not implemented in in this older version.

Thank you so much for your time and effort!

bubnikv, 2017-08-04:
You may try the prusa3d fork, where the G-code generator has been
rewritten. I hope the tool change will work as expected.

Jasper2991, 2017-08-07:
Thank you for your suggestion, 
i checked the prusa3d fork and encountered the same issue. 

I continued my search for a solution with 1.3.0-dev.
During my testing i did the following, Instead of using two objects with different extruders i generated a single object and added another part with a modifier.
[image]

This resulted in a single object with two parts, each using a different extruder.

with this (simple) method the generated G-code dos not contain unnecessary custom layer change G-code and my issue is solved. 

However, it might be wise to look in to this because it seems to be not possible to have different objects  and not having unwanted layer change G-code. even when using only one extruder.