slic3r/Slic3r #4067: Drag STL file into dock icon doesn't import file

qubist, 2017-07-21:
### Version
_1.3.0-dev_

### Operating system type + version
_macOS Sierra 10.12.5_

### Behavior
* _Dragging an STL file to the Slic3r dock icon opens Slic3r but doesn't import the file_
* _Drag an STL file to the Slic3r dock icon_
* _Slic3r opens and the "processing input file" window is visible followed by the file being imported._
* _Slic3r opens._

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Samir55, 2017-07-21:
I think it should be implemented like the one in this [commit](https://github.com/pkulchenko/ZeroBraneStudio/commit/f3168450a0a8a5467cc150e4e108e3bfed2626cc) using the MacNewFile() and MacOpenFile() funcntions in [WxApp class ](http://docs.wxwidgets.org/trunk/classwx_app.html)

alexrj, 2017-07-21:
Unfortunately we can't do this until we move to full C++. I already tried, and I wasn't able to catch MacOpenFile() through wxPerl.