slic3r/Slic3r #4068: Example of unsupported parts.

jaggzh, 2017-07-22:
### Version
1.2.9-627-g7d2884bc

### Operating system type + version
Debian Testing

### Behavior
Part of object is unsupported and also does not get raft underneath.

[noseguard.zip](https://github.com/alexrj/Slic3r/files/1167018/noseguard.zip)
[image]

jaggzh, 2017-07-22:
Also, it seems confused by the left edge of the object there, perhaps from the thinness. It generates a protrusion which does not match the rest of the vertical contour (the little yellow separated path at the left tip of the circled blue paths in the image).

freiser77, 2017-07-22:
see https://github.com/alexrj/Slic3r/issues/4054 also.