slic3r/Slic3r #4070: Reduce speed option doesn't apply to external perimeters

Maxwellfire, 2017-07-23:
### Version
Latest 1.3.0-Dev
Slic3r-master.2017.06.24.1248.13f121e.64bit

### Operating system type + version
Window ten

### Behavior
Expected results:

1. Set "Slow down if layer print time is below:" to 10 seconds 
2. Slice a part where the layer time is less than 10 seconds.
3. All print move speeds will be scaled together so that the total layer time is <10 seconds

Actual results:
All print moves except for external perimeters are slowed. External perimeters are printed at normal speed.

This leads to issues with the extruder switching from printing very slowly to printing very fast.

#### STL/Config (.ZIP) where problem occurs

[config.txt](https://github.com/alexrj/Slic3r/files/1167807/config.txt)

Maxwellfire, 2017-07-23:
I looked in the code, and this appears to be done deliberately. 
`# check that all layers have at least one unaltered external perimeter speed
    my $external = all { $_ > 0 } values %layer_external;
    ok $external, 'slowdown_below_layer_time does not alter external perimeters';`
What was the reasoning behind this?

bubnikv, 2017-07-23:
To keep the print quality of the outer shell consistent, independent of the
layer time.

Maxwellfire, 2017-07-23:
Okay. For me at least, quality depends more on smooth/slow transitions between speeds than on any absolute speed for a particular feature. I'll try the pressure advance option, and see if that helps. As it is it's completely ruining the outer surface :/.

bubnikv, 2017-07-23:
I agree with you that for the bowden setups the "new" way of slowing down the internal perimeters only may not be the best idea.

Maxwellfire, 2017-07-23:
Would it be useful then for me to add a print setting called something like "slowdown external perimeters" that when checked, always slows external perimeters, and then submit a pull request?

jmgiacalone, 2017-08-09:
I would prefer if the slowdown affected only infill and all perimeters were printed at the specified speed. I can imagine that for different objects people may have different preferences though. Perhaps it would make most sense to allow the user to select which features (infill, perimeters, external perimeters) are used for the slowdown?

bubnikv, 2017-10-05:
> Perhaps it would make most sense to allow the user to select which features (infill, perimeters, external perimeters) are used for the slowdown?

It is really about the external surfaces only. For example, changing the print speed changes surface finish of the print. I think the fast prints with PLA are more matte, the slow prints are more glossy. If you let Slic3r slow down on external perimeters, that part of a model will become visually different.

jmgiacalone, 2017-10-05:
I understand the impact on the surface finish, but I do think that whether they are affected by the slowdown or not, inner and external perimeters should be printed at the same speed as each other.

tcm0116, 2017-10-07:
I use the slow down feature to combat curling at the top of a part that is smaller than the rest. I would prefer that the surface finish be a little different as opposed to the top of the part failing due to it being printed too quickly. If I was concerned about surface finish, I would just slow the entire print down.

m-mccormick, 2017-10-07:
The issue that I'm running into is that my printer will no longer print small cross sectional areas with the "new" method. Period. My print quickly turns into the flying spaghetti monster, whereas version 1.34.1 didn't have an issue in this arena. Now, keep in mind that my printer is about as cheap as they come in terms of both cost and quality, but I'd like to think that eliminating some low end printers from being supported is less desirable than surface finish defects. I think it would be more beneficial to add more control options rather than choose one or the other, but since I am biased as a person who loves a ton of cusomizability I don't want to ask for the moon. Also, thank you very much to the people working on this.

hyperair, 2018-01-11:
I think this bug was introduced by this feature: https://github.com/alexrj/Slic3r/issues/2796