slic3r/Slic3r #4071: [feature request] variable linear advance settings per feature

curiouspl2, 2017-07-27:
Hello.
I test linear advance in marlin and it works _great_.
there is a problem though with printing very fast features with high accel and jerk values like zig-zag infill.
in this case geared extruders are too slow to cope with linear advance and simply drop steps. 

one of solutions is simply to slow down and reduce jerk and accel, but this causes prints with massive internal infill to print too slow. 

better solution would be to simply change linear advance setting like currently is done for acceleration settings - that is separate linear advance K would be present for infill, solid top infill , external and internal walls etc.

lordofhyphens, 2017-07-27:
This is probably too Marlin-specific to warrant addition here as written,
might be useful for some expansion of generic custom gcode attachments to
different print moves.

curiouspl2, 2017-07-27:
It could be solved once and for all along with adding jerk settings (like cura has since over half year and which are pretty useful) by just adding input boxes for custom gcode for each feature in advanced tab.

This way one can implement whatever one wishes, even turning on/off fans and changing colours (f.e. switching LED lights to red when printing infill and blue while printing external perimeters)
I guess also people hacking slic3r for CNC stuff like laser engraving will get all happy over such possibility  - this will allow for all possible hacks like changing spindle speeds, turning on coolants etc. etc.

lordofhyphens, 2017-08-03:
@curiouspl2 I was pretty sure that I said that ;) 

The UI as it is now, however, would be a colossal mess for adding custom gcode boxes for everything (pre-and post).