slic3r/Slic3r #4072: [feature request] Retraction status at beginning of print

joris57, 2017-07-29:
Slic3r expects the nozzle to be in a 'primed' state (as opposed to a 'retracted' state) at the beginning of a print (and, hence, does not start the first layer with an 'unretract' of the filament).

For a single extruder print, his does not cause much of an issue, but for multi-extruder, it would be nice if the start custom gcode could leave the printer in a 'retracted' state (to prevent oozing) and to start the first layer (for a given extruder) with an 'unretract' (using the toolchange settings for consistency reasons). This would make the logic for the first layer consistent with subsequent layers that also start with an 'unretract'.

For compatibility reasons, this behavior could be made optional.

lordofhyphens, 2017-08-03:
Could you describe the problem you are trying to solve in more detail please? Perhaps with a screenshot of the problematic area and a copy of the configuration + model that shows the undesired behavior?

joris57, 2017-08-04:
Thanks for looking into this. Let me try to be more clear. See also the attached image.

[image]

Background:
1) Dual extruder (2 in/2 out) print with an ooze shield.
2) Ooze shield is kept as thin as possible (1 loop) to avoid wasting plastic.
3) Model is slice with Slic3r-master.2017.06.24.1248.13f121e.32bit
4) Gcode is visualized with a competing slicer as Slic3r does not color-code the toolhead in its preview.

Observations:
1) The 'blue' extruder is nowhere properly purged.With a 1 loop wide ooze shield, the risk is very high that the ooze shield will no properly adhere and get knocked loose from the build plate.
2) The 'green' extruder is not purged at all.

Possible solution (by means of the start gcode):
1) Purge both extruders thru the printing of an 'intro line' at the edge of the printbed.
2) Do a retraction, to prevent excessive oozing of the inactive nozzle.

The retraction at the end of the start code of course means that the filament needs to be unretracted before starting to print. This unretract happens for every use of the extruder, EXCEPT FOR THE FIRST USE.

My suggestion is make it possible to specify that also for the first use of the extruder, an unretract needs to be done.

Let me know if you need further clarification or you have any other feedback on this issue.

joris57, 2020-04-26:
Cleaning up my list of open issues. This has become obsolete for me, so closing.

closed 2020-04-26 by joris57