slic3r/Slic3r #4073: [feature request] Time estimate for print

hyperair, 2017-07-30:
### Version

1.3.0-dev (98fca909)

### Operating system type + version
Ubuntu 17.04

### Behavior

After slicing and exporting to G-code, material consumption and price is shown, but not ETA. I think a rough theoretical ETA should be shown as well, as the current workflow for figuring out an ETA typically involves either using Octoprint, gcode.ws, or some other G-code analyzer to estimate the ETA.

lordofhyphens, 2017-07-30:
Dupe, actually. ;)

Also there is an optimistic estimator that is in libslic3r, I don't
remember if @alexrj put it in the UI.

lordofhyphens, 2017-08-01:
#3747 #1766 

The original PR did include, IIRC, an addition to the slice time box.

thiagopeixoto16, 2019-03-02:
@lordofhyphens How it is going? Where I can track this progress cause I find 3 issues and 2 pull requests and I don't know if you require testing... I also have some ideas to contribute but I really don't know where to inform :)

lordofhyphens, 2019-03-02:
@thiagopeixoto16 The PR is closed because @alranel implemented it.

lordofhyphens, 2019-03-02:
@thiagopeixoto16 Closing this issue because it is a duplicate of #1766

closed 2019-03-02 by lordofhyphens