slic3r/Slic3r #4075: Cut upper part preview incorrect

0xc0ffea, 2017-07-30:
### Version
Slic3r-master.2017.06.24.1248.13f121e.64bit

### Operating system type + version
Windows 10 64bit

### Behavior
The cut preview for the Upper part doesn't show the cut indicator plane position correctly when preview is enabled. Affects X,Y & Z, Output of cut is unaffected.

See attached video.
[2017-07-30_10-57-56.zip](https://github.com/alexrj/Slic3r/files/1185762/2017-07-30_10-57-56.zip)

dartrax, 2018-01-17:
I can confirm that on Win 7 64bit with Slic3r-master.2018.01.08.1579.19ea7a5.64bit
In my case, it occurs when "preview" and "upper part" is enabled, "lower part" is disabled.
[image]

dartrax, 2018-02-14:
SlicerPE does not have this issue.

lordofhyphens, 2018-02-14:
The Prusa fork has a different 3d rendering engine than the main one.