slic3r/Slic3r #4078: Top solid infill all over the surface

obor00, 2017-08-03:
### Version 1.3.0-dev
### Operating system Windows 8
### The top solid infill cover all the surface even if there are some others aera to print above
Infill = 25%
Top solid layer: 3
Infill pattern: any

Problem is that it changes the surface aspect outside the box at the level of the top solid surface.

On picture: top = slic3R, bottom = cura
[image]

#### STL/Config (.ZIP) where problem occurs
[Slic3r_config_bundle-taranTitan.ini.txt](https://github.com/alexrj/Slic3r/files/1198326/Slic3r_config_bundle-taranTitan.ini.txt)

lordofhyphens, 2017-08-03:
Need a copy of the model and the exported config, not the bundle.

lordofhyphens, 2017-08-03:
I'm not entirely sure here what you mean by "changes the surface aspect". It's internal fill for the internal areas and generally nobody cares what goes there. 

I need a copy of the exact configuration you used (Ctrl+E to export configs) along with a copy of a model that shows the behavior. Zip the two together and it may be uploaded.

obor00, 2017-08-04:
Here is the model :
[boitier_switch_pwm_V3-1.STL.txt](https://github.com/alexrj/Slic3r/files/1199595/boitier_switch_pwm_V3-1.STL.txt)
Config:
[config.ini.txt](https://github.com/alexrj/Slic3r/files/1199599/config.ini.txt)

See the line effect on the outside of the box:
[image]
[image]

obor00, 2017-08-07:
Do you have enough info ?

lordofhyphens, 2017-08-07:
That's what I asked for. I just can't get to every investigation
immediately all the time.

thierryzoller, 2020-04-18:
Wondered about the very same in 2020, where are we with this ?