slic3r/Slic3r #4081: Compiling error perl5.26 - MinGW-w64 on Windows XP SP3

redPrint7, 2017-08-07:
### Version
1.30 dev source  today

### Operating system type + version
Win XP SP3 / Windows 7 Ultimate 32 bit SP1 all updates

### Behavior
perl Build.PL stops with the following error:
file \xs\src\boost\nowide\cenv.hpp
Error: cenv.hpp:104:41: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
in cenv.hpp line 104/41

 if I change from

    while(*key_end!='=' && key_end!='\0')
 to    
  while(*key_end!='=' && key_end!="\0") //changed '\0' to "\0"
  it works as expected.
Infos on build environment perl 5.26 compiled from source on WinXP SP3
Mingw-w64 v5.02 toolchain 

Problem could be reproduced on win 7 ultimate 32bit with the same toolchain - it just popped up on XP

Background: Since Prusas Slic3r-version is not (anymore) compiled for XP I have to do it for my own with free cross-platform tools I use for other projects. To find out where Errors/bugs come from I always compile Master and prusa Ed with the same toolchain. This "error" seems to belong to Master.
Except some probs with local::lib all compiled and worked as far I could test.

lordofhyphens, 2017-08-07:
We have a specific environment for building on Windows based on Strawberry
Perl, which is what our build servers use.

Boost nowide isn't our code either and you probably have strict settings in
your cflags?

I would report this up to the nowide author and patch it locally.

lordofhyphens, 2017-08-07:
The Prusa3d fork probably hasn't pulled that patch down from us yet either.

redPrint7, 2017-08-10:
NowIde owner had this problem as issue 14, but only as warning annoyance, added the issue with compiling stop. So maybe there is a permanent fix in future.
Thanks for your work

closed 2017-08-10 by redPrint7