slic3r/Slic3r #4082: Http Request on post-slicing script?

Mendes11, 2017-08-09:
Hi, I have an Android App called Price My Print (feel free to download it! ;D) which i'm developing a new feature that requires the slicers to communicate with a php script when they finish their slicing work sending basic infos like fillament lengh/wheight and Extimated Time. I would like to know if there is a way of doing this with slic3r by some post-slicing script feature or if it would be possible to pull request to the code something that if enabled by the user it does the communication. If yes for one of the two possibilities, how would be the best way of doing it?

Thanks in advance

lordofhyphens, 2017-08-09:
Post processing script. I think everything is available w/r/t time
estimates, etc as well

Mendes11, 2017-08-09:
@lordofhyphens  but could I make a POST HTTP request on this Post processing script?

I'm asking this because I'm thinking about the libs I could need to do this but i guess I can just put them together with the script and that's it, right?

lordofhyphens, 2017-08-09:
By default it is Perl (or some other executable), so you can do whatever
you want.

closed 2017-08-09 by Mendes11