slic3r/Slic3r #4085: GL_MULTISAMPLE and GL_ARRAY_BUFFER_ARB messages on failed launch

citybadger, 2017-08-12:
### Version
1.2.9-1120-g13f121e3

### Operating system type + version
Debian testing:

Linux minor7th 4.11.0-1-rt-amd64 #1 SMP PREEMPT RT Debian 4.11.6-1 (2017-06-19) x86_64 GNU/Linux

(have tried using the standard non-realtime kernel with same result)

### Behavior
Error below when run. 

```
citybadger@minor7th:~/sourceCode/Slic3r$ ./slic3r.pl 
Your vendor has not defined OpenGL macro GL_MULTISAMPLE, used at /home/citybadger/sourceCode/Slic3r/lib/Slic3r/GUI/3DScene.pm line 747.
```
If I comment out that line, and the other instance of GL_MULTISAMPLE that follows, I get this message:

```
Your vendor has not defined OpenGL macro GL_ARRAY_BUFFER_ARB, used at /home/citybadger/sourceCode/Slic3r/lib/Slic3r/GUI/3DScene.pm line 914.
```
If I disable the those lines by changing the ifs surrounding that line and the two subsequent lines to if(0), ./slic3r.pl launches sucessfully. 
 
Output of  `perl Build.PL"

```
Attempting to create directory /home/citybadger/sourceCode/Slic3r2/Slic3r/local-lib
--> Working on App::cpanminus
Fetching http://www.cpan.org/authors/id/M/MI/MIYAGAWA/App-cpanminus-1.7043.tar.gz ... OK
Configuring App-cpanminus-1.7043 ... OK
Building and testing App-cpanminus-1.7043 ... OK
Successfully installed App-cpanminus-1.7043 (upgraded from 1.7042)
1 distribution installed
Class::XSAccessor is up to date. (1.19)
--> Working on Devel::CheckLib
Fetching http://www.cpan.org/authors/id/M/MA/MATTN/Devel-CheckLib-1.11.tar.gz ... OK
Configuring Devel-CheckLib-1.11 ... OK
==> Found dependencies: Mock::Config, IO::CaptureOutput
--> Working on Mock::Config
Fetching http://www.cpan.org/authors/id/R/RU/RURBAN/Mock-Config-0.03.tar.gz ... OK
Configuring Mock-Config-0.03 ... OK
Building and testing Mock-Config-0.03 ... OK
Successfully installed Mock-Config-0.03
--> Working on IO::CaptureOutput
Fetching http://www.cpan.org/authors/id/D/DA/DAGOLDEN/IO-CaptureOutput-1.1104.tar.gz ... OK
Configuring IO-CaptureOutput-1.1104 ... OK
Building and testing IO-CaptureOutput-1.1104 ... OK
Successfully installed IO-CaptureOutput-1.1104
Building and testing Devel-CheckLib-1.11 ... OK
Successfully installed Devel-CheckLib-1.11
3 distributions installed
--> Working on Encode
Fetching http://www.cpan.org/authors/id/D/DA/DANKOGAI/Encode-2.92.tar.gz ... OK
Configuring Encode-2.92 ... OK
Building and testing Encode-2.92 ... OK
Successfully installed Encode-2.92 (upgraded from 2.88)
1 distribution installed
Encode::Locale is up to date. (1.05)
--> Working on ExtUtils::CppGuess
Fetching http://www.cpan.org/authors/id/D/DA/DAVIDO/ExtUtils-CppGuess-0.11.tar.gz ... OK
Configuring ExtUtils-CppGuess-0.11 ... OK
Building and testing ExtUtils-CppGuess-0.11 ... OK
Successfully installed ExtUtils-CppGuess-0.11
1 distribution installed
--> Working on ExtUtils::MakeMaker
Fetching http://www.cpan.org/authors/id/B/BI/BINGOS/ExtUtils-MakeMaker-7.30.tar.gz ... OK
Configuring ExtUtils-MakeMaker-7.30 ... OK
Building and testing ExtUtils-MakeMaker-7.30 ... OK
Successfully installed ExtUtils-MakeMaker-7.30 (upgraded from 7.24)
[... 525 lines cut ...]
t/gcode.t ...................... ok     
t/geometry.t ................... ok     
t/layers.t ..................... ok   
t/loops.t ...................... ok   
t/multi.t ...................... ok     
t/perimeters.t ................. ok     
t/polyclip.t ................... ok     
t/pressure.t ................... ok   
t/print.t ...................... ok   
t/retraction.t ................. ok     
t/shells.t ..................... ok     
t/skirt_brim.t ................. ok   
t/slice.t ...................... skipped: temporarily disabled
t/speed.t ...................... ok   
t/support.t .................... ok     
t/svg.t ........................ ok   
t/thin.t ....................... ok     
t/threads.t .................... ok   
t/vibrationlimit.t ............. ok   
All tests successful.
Files=34, Tests=580, 47 wallclock secs ( 0.14 usr  0.02 sys + 50.23 cusr  0.98 csys = 51.37 CPU)
Result: PASS
If you also want to use the GUI you can now run `perl Build.PL --gui` to install the required modules.

```

citybadger, 2017-08-12:
Update: this is also happening (same message but a different line number)  with the version installed with apt-get from the Debian archives. I don't keep an mixed system or anything, but perhaps this is a conflict with a recently upgraded library or a Debian specific problem.

stylesuxx, 2017-08-12:
I seem to have exactly the same problem with slic3r from the Repos, on Debian testing:

`Your vendor has not defined OpenGL macro GL_MULTISAMPLE, used at /usr/lib/slic3r/Slic3r/GUI/3DScene.pm line 619.
`

lordofhyphens, 2017-08-13:
Ugh, this is probably POGL related, or your driver version?

citybadger, 2017-08-13:
I'm pretty sure both Debian's and my self-compiled version were working until recently - perhaps as recently as last week. Looking in dpkg.log, I see that libopengl-perl and slic3r were upgraded:

2017-08-07 17:37:50 upgrade libopengl-perl:amd64 0.6704+dfsg-2+b2 0.6704+dfsg-2+b4
2017-08-07 17:39:32 upgrade slic3r:amd64 1.2.9+dfsg-6 1.2.9+dfsg-6+b1

lordofhyphens, 2017-08-13:
Then it is probably opengl perl, I don't know what the debian package would
need.

Use the appimage in the meantime.

lordofhyphens, 2017-08-13:
I know that POGL 0.70 and 0.67 should be working  (I have 0.70 and had to
fix something in slic3r because of it months ago).

lordofhyphens, 2017-09-03:
@citybadger I'm suspecting it's your system right now; I just upgraded my test system to Stretch and I haven't run into that problem. 

Usually when some macro isn't defined your OpenGL libraries are weird or the driver is old. I can probably work around it though.

citybadger, 2017-09-03:
I'm on testing, aka Buster. I believe the libraries were upgraded after
Stretch became stable, so my libraries are newer than yours. If this is a
bug in the new libraries, I expect it will eventually be corrected. If it's
a feature, then this problem will grow as more and more distributions
upgrade to the new libraries.

lordofhyphens, 2017-09-03:
Any problems with the AppImage build?

citybadger, 2017-09-03:
The AppImage build runs on my system.

citybadger, 2017-09-03:
The library versions in the AppImage are older than on my system. I don't
know what to look for, but for example libwx_gtk2u_gl-3.0.so.0.2.0 is in
the AppImage and libwx_gtk2u_gl-3.0.so.0.3.0 is on my system.

If you could tell me which libraries are the likely source of this problem
I could compare them.

rockstorm101, 2017-09-04:
I'm running Debian Buster and I'm experiencing the very same error.

In case it helps, @citybadger, according to [this question](https://stackoverflow.com/questions/4207506/where-is-gl-multisample-defined), _GL_MULTISAMPLE_ is defined:
 * Within 'glext.h', which, in Debian, is provided by, among some others, the "mesa-common-dev" and "libgtkglext1-dev" packages . However Slic3r doesn't seem to depend on them directly.
 * Or in 'glew.h', which is provided by the "libglew-dev" package.

lordofhyphens, 2017-09-04:
Does it work if any of those packages are installed?

rockstorm101, 2017-09-06:
Installing them ("mesa-common-dev", "libgtkglext1-dev" nor "libglew-dev") does not solve the problem.

Judging by the error message it would seem _GL_MULTISAMPLE_ is either not defined or Slic3r is somehow unable to find the definition, but neither of them make much sense to me. I'm certainly lost here.

lordofhyphens, 2017-09-06:
This is down in the POGL libraries that Slic3r depends on, which makes it
extra hard to test/diagnose. That it works in Stretch but not in Buster
makes me wonder whether or not the opengl libraries have a bug in them.

rockstorm101, 2017-09-06:
If you have any idea of what I could test/diagnose to try to narrow down the problem I would happily do it.

lordofhyphens, 2017-09-06:
Someone wrote up the same issue against the slic3r-prusa package on Debian. 

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=872273
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=872258

This seems relevant.
https://stackoverflow.com/questions/4207506/where-is-gl-multisample-defined

Can you replace the instances of GL_MULTISAMPLE with GL_MULTISAMPLE_ARB ?

lordofhyphens, 2017-09-06:
Alternatively, do you have glew installed (might be relevant): 
https://packages.debian.org/source/buster/glew

russell, 2017-09-11:
Hi, I tried renaming the import from GL_MULTISAMPLE to GL_MULTISAMPLE_ARB, but it just erred on the new symbol.  

I tried installing the libglew-dev stuff as you suggested, but that didn't seem to work either. 

I have followed the advice in the Debian bug report and deleted the lines with GL_MULTISAMPLE on and it seems to work for now. :crossed_fingers: 

Do you know if POGL is reading those symbols out of the header files.  Perhaps that constant doesn't exist in the latest mesagl drivers or something?  It looks like one of the Debian bugs is against mesa.

lordofhyphens, 2017-09-11:
It looked like to me that the debian bug against mesa was commented on as
"wtf do you expect us to do".

What POGL does is that it translates it into something else. I've seen it
dying on other constants that usually were tied to driver/libgl problems.

Prusa3D's fork of Slic3r has the same problem.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-09-11:
I am glad to hear that the AppImage works; so everyone isn't out in the cold.

bubnikv, 2017-09-12:
> Prusa3D's fork of Slic3r has the same problem.

I wonder why they did not open an issue at our git.

Anyway, the correct solution is to extend our "can_multisample" check

    my $can_multisample =
        Wx::wxVERSION >= 3.000003 &&
        defined Wx::GLCanvas->can('WX_GL_SAMPLE_BUFFERS') &&
        defined Wx::GLCanvas->can('WX_GL_SAMPLES');
with

        defined Wx::GLCanvas->can('multisample')

then to store the "can_multisample" at the class and check it before
each glEnable/glDisable(GL_MULTISAMPLE);

bubnikv, 2017-09-12:
> Anyway, the correct solution is to extend our "can_multisample" check

Sorry, that was not a working solution.

bubnikv, 2017-09-12:
I have hopefully fixed it with
https://github.com/prusa3d/Slic3r/commit/888a904c9b484289318737ec583ac0794dfa84f7
and
https://github.com/prusa3d/Slic3r/commit/6d4ec5c989851e442d58f896d4b0324129e7514a

onitake, 2017-09-29:
Just for the record: I'm pretty sure installing the -dev packages won't resolve anything. They are only used during package building (in this case, POGL) and won't contribute anything after.

It's very likely that the POGL symbol scanner failed to detect the mentioned OpenGL symbols.
Can you try building libopengl-perl on your system (perhaps with additional -dev packages) and check if you see anything suspicious?
If it finds the symbols, try installing the self-built library and see if Slic3r works. If it does, you should report a bug against libopengl-perl on Debian. If not, it's likely a bug in POGL itself and you should report it on their tracker.

lordofhyphens, 2017-10-23:
@bubnikv I just noticed your proposed fix and cherry-picked it. 

@onitake https://github.com/alexrj/Slic3r/pull/4167 can you see if that fixes it for you?

NeonHorizon, 2017-10-28:
I've updated to Kubuntu 17.10 from 17.04 and hit this same issue with the 1.2.9 version in Ubuntu's repositories.
For me what I did was download Slic3r from here https://dl.slic3r.org/linux/ and its fine.
Not sure if that helps anyone?

hyperair, 2017-11-05:
@lordofhyphens I had to hack `HAS_VBO` to test for the presence of `GL_ARRAY_BUFFER_ARB` as well to make this work on Ubuntu 17.10:

```diff
diff --git a/lib/Slic3r/GUI/3DScene.pm b/lib/Slic3r/GUI/3DScene.pm
index 72e7d28e..c5d903f3 100644
--- a/lib/Slic3r/GUI/3DScene.pm
+++ b/lib/Slic3r/GUI/3DScene.pm
@@ -50,7 +50,7 @@ use constant PI             => 3.1415927;
 
 # Constant to determine if Vertex Buffer objects are used to draw
 # bed grid and the cut plane for object separation.
-use constant HAS_VBO        => eval { glGenBuffersARB_p(0); 1 };
+use constant HAS_VBO        => eval { glGenBuffersARB_p(0); GL_ARRAY_BUFFER_ARB; 1 };
 
 
 # phi / theta angles to orient the camera.
```

onitake, 2017-11-06:
I wonder why these features are so problematic.
Practically every GPU and OGL library has support for them these days???

On the other hand, the ARB variants are pre-OpenGL2 leftovers.
One should be able to use the non-ARB variants without problems, no?

bubnikv, 2017-11-06:
The Perl OpenGL module does not support OpenGL 2.0 and higher, so the
OpenGL 2.0 features are only accessible through the extensions by the Perl
OpenGL module.

The Prusa3D fork ditched most of the Perl OpenGL and the OpenGL 2.0 calls
are directly executed from a C++ code.

Patola, 2017-11-24:
So... how to fix this problem? I have it too now.

wseverin, 2017-11-28:
+1 affected. Ubuntu 17.10. Thankfully the .AppImage seems to work.

arofarn, 2017-12-03:
I have the same problem on Kubuntu 17.10, with slic3r (1.2.9) and slic3r-prusa (1.33.8-dev) from the ubuntu repositories.

I've downloaded Slic3r from the download page ( http://slic3r.org/download ) and it works out the box without any problem. Same with Slic3r Prusa Ed...
So I think the problem is with some packages in ubuntu repositories....

hyperair, 2017-12-05:
On Sun, Dec 03, 2017 at 06:18:43PM +0000, Arofarn wrote:
> I have the same problem on Kubuntu 17.10, with slic3r (1.2.9) and slic3r-prusa (1.33.8-dev) from the ubuntu repositories.
>
> I've downloaded Slic3r from the download page ( http://slic3r.org/download [...]

I just uploaded a fix for this at
https://bugs.launchpad.net/ubuntu/artful/+source/slic3r/+bug/1726103

Could you help test this please?

-- 
Kind regards,
Loong Jin

citybadger, 2017-12-05:
Currently I still have this problem when compiling from the source (from
github). But the package from the Debian repositories is working.

lordofhyphens, 2017-12-05:
Debian package repo is *old*.

hyperair, 2017-12-14:
That's because your latest stable release is *old*.

GilesBathgate, 2017-12-14:
This also happened to me after upgrading to Ubuntu 17.10 artful. @hyperair 's fix to HAS_VBO worked, then I just commented out the two line containing GL_MULTISAMPLE. There are two more occourances of GL_MULTISAMPLE in master not covered by bubnikv's commit here: https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/3DScene.pm#L747 and here: https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/3DScene.pm#L772

lordofhyphens, 2017-12-15:
Thanks @GilesBathgate

PR #4230 (2017-12-15): Fixes the GL_MULTISAMPLE and GL_ARRAY_BUFFER_ARB fail to launch
fixes #4085
closed 2017-12-15 by lordofhyphens