slic3r/Slic3r #4089: Installing cpanminus in Slic3r Perl on 64-bit Windows 10

calesh, 2017-08-18:
### Version
Master (pulled 2017 Aug 14)

### Windows 10 Home + 1607 (64-bit)

### Behavior

I am trying to set up an environment to compile Slic3r on a 64-bit Windows 10 machine with the hopes of improving support material. I have downloaded the custom Slic3r Perl <https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#slic3r-perl>, but when I try to build, I receive an error message saying cpanminus needs to be installed. I tried all of the methods to install cpanm that were listed in the error log. What can I do to install cpanminus with my set up?

```
C:\Users\David\Documents\3D_Printer\Slic3r>perl Build.PL
cpanm was not found. Please install it before running this script.

There are several ways to install cpanm, try one of these:

    apt-get install cpanminus
    curl -L http://cpanmin.us | perl - --sudo App::cpanminus
    cpan App::cpanminus

If it is installed in a non-standard location you can do:

    CPANM=/path/to/cpanm perl Build.PL

C:\Users\David\Documents\3D_Printer\Slic3r>
```

<p>I tried cpan App:cpanminus (call this method 3)</p>

```
C:\Users\David\Documents\3D_Printer\Slic3r>cpan App::cpanminus
Loading internal null logger. Install Log::Log4perl for logging messages
CPAN: Storable loaded ok (v2.56_01)
CPAN: LWP::UserAgent loaded ok (v6.24)
CPAN: Time::HiRes loaded ok (v1.9741)
Fetching with LWP:
http://www.cpan.org/authors/01mailrc.txt.gz
Reading '\c\Users\David\.cpan\sources\authors\01mailrc.txt.gz'
CPAN: Compress::Zlib loaded ok (v2.074)
............................................................................DONE
Fetching with LWP:
http://www.cpan.org/modules/02packages.details.txt.gz
Reading '\c\Users\David\.cpan\sources\modules\02packages.details.txt.gz'
  Database was generated on Thu, 17 Aug 2017 21:53:52 GMT
............................................................................DONE
Fetching with LWP:
http://www.cpan.org/modules/03modlist.data.gz
Reading '\c\Users\David\.cpan\sources\modules\03modlist.data.gz'
DONE
Writing \c\Users\David\.cpan\Metadata
Running install for module 'App::cpanminus'
Fetching with LWP:
http://www.cpan.org/authors/id/M/MI/MIYAGAWA/App-cpanminus-1.7043.tar.gz
CPAN: Digest::SHA loaded ok (v5.96)
Fetching with LWP:
http://www.cpan.org/authors/id/M/MI/MIYAGAWA/CHECKSUMS
Checksum for \c\Users\David\.cpan\sources\authors\id\M\MI\MIYAGAWA\App-cpanminus-1.7043.tar.gz ok
The system cannot find the path specified.
Uncompressed \c\Users\David\.cpan\sources\authors\id\M\MI\MIYAGAWA\App-cpanminus-1.7043.tar.gz successfully
Using Tar:/usr/bin/tar xf "App-cpanminus-1.7043.tar":
The system cannot find the path specified.
Couldn't untar App-cpanminus-1.7043.tar: child exited with value 1
'YAML' not installed, will not store persistent state
  MIYAGAWA/App-cpanminus-1.7043.tar.gz
  Had problems unarchiving. Please build manually

C:\Users\David\Documents\3D_Printer\Slic3r>
```

<p>I also tried, (call this method 1)</p>

```
C:\Users\David\Documents\3D_Printer\Slic3r>apt-get install cpanminus
'apt-get' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\David\Documents\3D_Printer\Slic3r>
```

And I tried, curl... (call this method 2)

```
C:\Users\David\Documents\3D_Printer\Slic3r>curl -L http://cpanmin.us | perl - --sudo App::cpanminus
'curl' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\David\Documents\3D_Printer\Slic3r>
```

<p>In method 3, I noticed that it is trying to install in \c\Users\... which creates a directory called 'c' off the root of drive C: (i.e. C:\c\Users...etc.). Thus, I see a Unix/DOS naming problem.</p>

<p>How can I install cpanm?</p>

cangelis, 2017-08-18:
Which perl dist and shell are you working on?

cangelis, 2017-08-18:
Try Strawberry Perl. 

Download this: http://strawberryperl.com/download/5.24.1.1/strawberry-perl-5.24.1.1-32bit-portable.zip

Unzip it & run `portableshell.bat`

lordofhyphens, 2017-08-18:
> Try Strawberry Perl.
>
> Download this: http://strawberryperl.com/download/5.24.1.1/strawberry-
> perl-5.24.1.1-32bit-portable.zip
>
> Unzip it & run portableshell.bat

Original poster was using 64bit, not 32.

I suspect that the PATH variable is not set up properly.

You can extract the other archive mentioned in the wiki.

lordofhyphens, 2017-08-18:
@cangelis the Slic3r perl is a version of the Strawberry perl distribution (I was the one who made the modifications).

@calesh Ensure that your PATH  variable has the correct items in it 

https://github.com/alexrj/Slic3r/blob/master/package/win/appveyor_preinstall.ps1 is what the build server uses to get most of the env set up right. 

https://github.com/alexrj/Slic3r/blob/master/appveyor.yml is what the build server runs when it starts.

`apt-get`  is a Debian/Ubuntu specific command (it's their package manager).

cangelis, 2017-08-18:
@lordofhyphens good to know that. 👍 

Btw, running`portableshell.bat` instead of `cmd` must solve the problems.

calesh, 2017-08-18:
@lordofhyphens Thank you for your prompt response. 

1. My PATH variable and cpan configuration are attached below. Sorry, I don't know what the "correct items" in the PATH should be. The cpan utility appears to search in Unix-like paths (/usr/bin/tar) i.e. no drive letter. I think that this is incompatible with the Windows-style notation (C:\Users\...). I don't understand how adding things to the PATH variable will enable cpan to work. How can I resolve this problem?

1. When you say "extract the other archive", which archive are you referring to?

1. Just to confirm, I was under the impression that I **must** use only your custom Slicer Perl if I am on a 64-bit machine (issue #3815) **and** I want the 3D Preview capability (cannot use Citrus Perl).  Is this correct?

Sorry for being so dense; I'm new to GitHub and perl. Thanks for your help.

[path_and_cpanm_conf.txt](https://github.com/alexrj/Slic3r/files/1235716/path_and_cpanm_conf.txt)

lordofhyphens, 2017-10-02:
It looks like to me that your PATH is probably set up right. 

You **must** use Slic3r Perl if you want 64-bit Slic3r.. Otherwise you can't compile Slic3r because the compiler is too old (minimum compiler is gcc 4.9 and Strawberry Perl's gcc-4.9 uses an exception model that causes application freezes). 

If you are fine with 32bit Slic3r you can use the normal Strawberry Perl build.

If you set the environment variable $ARCH=64bit, rename the references to c:\users\appveyor to somewhere else, and trim out the dangerous bits from  https://github.com/alexrj/Slic3r/blob/master/package/win/appveyor_preinstall.ps1 you can run it to set up your system. 

these other environment variables may be of use to you. 
````
  LDLOADLIBS: -lstdc++
  SLIC3R_STATIC: 1
  SLIC3R_VERSION: 1.3.0
  BOOST_DIR: C:\dev\boost_1_63_0
  WXDIR: C:\dev\wxwidgets
  WXSHARED: SHARED=0
  FORCE_WX_BUILD: 0
  FORCE_BOOST_REINSTALL: 0
````

mikko70, 2017-11-15:
You need copy dmake in C:\wperl64\bin
http://search.cpan.org/~pczerkas/dmake-4.12-cvs-20090821-PCZERKAS/

if you run files as administrator, install happen.

[wperl64.zip](https://github.com/alexrj/Slic3r/files/1476119/wperl64.zip)

N4rki, 2017-12-08:
I have the exact same issue, tried all those things but still couldn't get it to work. @calesh Did you find a solution in the meantime?

I get this this error with all methods: 
"Loading internal null logger. Install Log::Log4perl for logging messages
CPAN: Storable loaded ok (v2.56_01)
Reading '\c\Users\jcj-vb\.cpan\Metadata'
  Database was generated on Fri, 08 Dec 2017 13:17:02 GMT
Running install for module 'App::cpanminus'
CPAN: Digest::SHA loaded ok (v5.96)
CPAN: Compress::Zlib loaded ok (v2.074)
Checksum for \c\Users\jcj-vb\.cpan\sources\authors\id\M\MI\MIYAGAWA\App-cpanminus-1.7043.tar.gz ok
Das System kann den angegebenen Pfad nicht finden.
Uncompressed \c\Users\jcj-vb\.cpan\sources\authors\id\M\MI\MIYAGAWA\App-cpanminus-1.7043.tar.gz successfully
Using Tar:/usr/bin/tar xf "App-cpanminus-1.7043.tar":
System couldn't find path.
Couldn't untar App-cpanminus-1.7043.tar: child exited with value 1
'YAML' not installed, will not store persistent state
  MIYAGAWA/App-cpanminus-1.7043.tar.gz
  Had problems unarchiving. Please build manually
"

Any other suggestions? And how CAN I build it manually? Buildung tar.gz files on Windows is new to me.
All help is much appreciated.

calesh, 2017-12-10:
@N4rki No, I have not yet found a solution.  (I have not yet tried the suggestion from mikko70 on 2017 Nov 15.)

@mikko70 Thanks for your suggestion.  I have been away from this project and have not tried yet.

lordofhyphens, 2017-12-10:
@calesh by the way Strawberry Perl 5.26 shouldn't have the exception bug, but I haven't tried building on that yet.

Building cpanm manually would involve downloading it (see source link at http://search.cpan.org/~miyagawa/App-cpanminus-1.7043/lib/App/cpanminus.pm), extracting it to a directory, and then running `perl Makefile.PL`. 

That should detect your system and generate an appropriate makefile. 

You'd then either run mingw-32-make or dmake (depending on which is available) with the "install" target.

robbycandra, 2018-05-20:
I use Strawberry Perl 5.24 32 bit and meet same problem.

My result  :
C:\Strawberry\perl\bin>cpan App::cpanminus

Loading internal null logger. Install Log::Log4perl for logging messages
CPAN: CPAN::SQLite loaded ok (v0.211)
Database was generated on Sun, 20 May 2018 03:34:57 GMT

Running install for module 'App::cpanminus'
CPAN: Digest::SHA loaded ok (v6.01)
CPAN: Compress::Zlib loaded ok (v2.081)
Checksum for C:\STRAWB~1\cpan\sources\authors\id\M\MI\MIYAGAWA\App-cpanminus-1.7044.tar.gz ok
CPAN: Archive::Tar loaded ok (v2.26)
CPAN: YAML::XS loaded ok (v0.69)
CPAN: CPAN::Meta::Requirements loaded ok (v2.140)
CPAN: Parse::CPAN::Meta loaded ok (v2.150010)
CPAN: CPAN::Meta loaded ok (v2.150010)
CPAN: Module::CoreList loaded ok (v5.20180414_24)
Configuring M/MI/MIYAGAWA/App-cpanminus-1.7044.tar.gz with Makefile.PL
Checking if your kit is complete...
Looks good
Generating a dmake-style Makefile
Writing Makefile for App::cpanminus
Writing MYMETA.yml and MYMETA.json
  MIYAGAWA/App-cpanminus-1.7044.tar.gz
  C:\Strawberry\perl\bin\perl.exe Makefile.PL -- OK
Running make for M/MI/MIYAGAWA/App-cpanminus-1.7044.tar.gz
cp lib/App/cpanminus/fatscript.pm blib\lib\App\cpanminus\fatscript.pm
cp lib/App/cpanminus.pm blib\lib\App\cpanminus.pm
"C:\Strawberry\perl\bin\perl.exe" -MExtUtils::Command -e cp -- bin/cpanm blib\script\cpanm
pl2bat.bat blib\script\cpanm
Can't execute /c/Strawberry/perl/bin/pl2bat.bat.
dmake.exe:  Error code 129, while making 'blib\script\cpanm'
dmake.exe:  'blib\script\cpanm' removed.
  MIYAGAWA/App-cpanminus-1.7044.tar.gz
  C:\STRAWB~1\c\bin\dmake.exe -- NOT OK
Stopping: 'install' failed for 'App::cpanminus'.

Please Help.
I already try Strawberry 64bit fail on same problem 
CitrusPerl, fail on different problem

robbycandra, 2018-05-20:
When i run perl build.pl :

D:\MyApps\Slicer\Slic3r>perl build.pl
cpanm was not found. Please install it before running this script.

There are several ways to install cpanm, try one of these:

    apt-get install cpanminus
    curl -L http://cpanmin.us | perl - --sudo App::cpanminus
    cpan App::cpanminus

If it is installed in a non-standard location you can do:

    CPANM=/path/to/cpanm perl Build.PL
------------------------------------------------
But ....

C:\Strawberry\perl\bin>perl -MApp::cpanminus -le "print 'ok'"
ok

C:\Strawberry\perl\bin>perl -le "print for grep { /(berry|perl)/ } split /;/, $ENV{PATH}"
C:\Strawberry\c\bin
C:\Strawberry\perl\site\bin
C:\Strawberry\perl\bin

C:\Strawberry\perl\bin>cpanm --version
cpanm (App::cpanminus) version 1.7042 (cpanm)
perl version 5.024004 (C:\Strawberry\perl\bin\perl.exe)

  %Config:
    archname=MSWin32-x86-multi-thread-64int
    installsitelib=C:\STRAWB~1\perl\site\lib
    installsitebin=C:\STRAWB~1\perl\site\bin
    sitearchexp=C:\STRAWB~1\perl\site\lib
    sitelibexp=C:\STRAWB~1\perl\site\lib
    vendorarch=C:\STRAWB~1\perl\vendor\lib
    vendorlibexp=C:\STRAWB~1\perl\vendor\lib
    archlibexp=C:\STRAWB~1\perl\lib
    privlibexp=C:\STRAWB~1\perl\lib
  %ENV:
  @INC:
    FatPacked::42640092=HASH(0x28aa2dc)
    C:/Strawberry/perl/site/lib
    C:/Strawberry/perl/vendor/lib
    C:/Strawberry/perl/lib/MSWin32-x86-multi-thread-64int
    C:/Strawberry/perl/lib
    .

robbycandra, 2018-05-20:
FInally. I can build it using @cangelis workaround. 

When I run "perl Build.PL" i have another problem

> Installing ./xs failed.

But after running 
'''
perl -MCPAN -e shell
CPAN> o conf make /usr/bin/make
'''
and re run perl Build.PL, everything is OK now.

closed 2019-01-21 by lordofhyphens