slic3r/Slic3r #4092: Feature request: Wipe standalone (no need retract and customizable length)

stepkurniawan, 2017-08-21:
### Slicer Version 1.2.9

### Operating system: Windows 10

### Behavior
 _Retraction options:  --wipe              Wipe the nozzle while doing a retraction (default: no)_

Currently, the --wipe **(wipe while retracting)** function is too magical for me. First, it only works while the retraction is also turned on. Second, I cannot tweak the length in which the wipe function is going. 
(I am using --wipe in the command line)

In my project, the blobs I'm having is quite severe. I want to just "brush it off" by wiping in to the rear. And since I am using a custom 3D Printer, the scale that the wipe is now have is not large enough. I guess 10 mm would be the best, but it is just my assumption. 

And I like to use the command line version. Something like "--wipe 10" to wipe 10mm in a reverse direction would be awesome.

I am new to Git, so if there is some suggestion please feel free to tell me. I also have difficulty in searching for the wipe function itself in the Git, can someone point it out? 

Thank you!

lordofhyphens, 2017-08-21:
I don't see how customizable length is generally possible, given that it is
moving into the infill to let ooze drop out where you won't see it.

I believe the control over that is in GCode.cpp (class Wipe) if you want to
mess around with it yourself.

It produces a path for the wipe motion.

lordofhyphens, 2017-08-21:
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/GCode.cpp#L145

The function tries to wipe along the entire retract move, which is how it gets its length.

stepkurniawan, 2017-08-21:
Is this true that if I were to increase it by 10x, the code that I can edit is at: 
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/GCode.cpp#L162

double wipe_dist = scale_(length / gcodegen.writer.extruder()->retract_speed() * wipe_speed) ***10**;

And what does it means that the length is a toolchange? 💃 
double length = toolchange
from:
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/GCode.cpp#L154

lordofhyphens, 2017-08-21:
> libslic3r/GCode.cpp#L162
> libslic3r/GCode.cpp#L154

1. Probably, although you may just want to add _scale(10) to get the output
in mm (I think), instead of multiplying by 10. Add a printf debug there to
see what the values look like.
2. Look at the next two lines; it's a ternary operator based on whether or
not this wipe is for a toolchange.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

stepkurniawan, 2017-08-21:
I... don't know how to run the code :P 
Is it the same as to building the Slic3r? If that so, I have to follow the instructions in the 
[How_to_build_Slic3r.txt](https://github.com/stepkurniawan/Slic3r/blob/master/doc/How_to_build_Slic3r.txt) then?

Samir55, 2017-08-21:
* Then you run those commands at the Slic3r repository folder
```Console
mkdir build
cd build
cmake ../src
make
then you get the executable slic3r in the build folder
```

Samir55, 2017-08-21:
The previous steps are much faster if you are changing in C++ code than running perl Build.pl

lordofhyphens, 2017-08-21:
https://github.com/alexrj/Slic3r/wiki should be the general reference for build instructions.

stepkurniawan, 2017-08-22:
@Samir55 I tried using cmd, but the result was: "the command cmake is either falsely written or cannot be found"

I'll follow the instruction to compile the slic3r from : https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows then, by installing the perl

stepkurniawan, 2017-08-25:
@lordofhyphens @Samir55 

Hey! Thanks for the help! I successfully build the Slic3r and such! It was an experience indeed :)
btw, there is an incompatibility issue for me to use Slicer 1.30, so I have to use the stable version 1.29, but I could not find the GCode.cpp there. Do you know where can I find the wipe function in the Slicer version 1.29?

lordofhyphens, 2017-08-25:
What is wrong? If you are using the correct Perl and compiler setup then it
should build.

I believe that wipe is still part of xs in 1.2.9 which means you would need
a source checkout of the tag if you go that route.

stepkurniawan, 2017-08-25:
The problem does not lies in the Slic3r, but rather in my post processor. 
I encounter the problem of :
in the Slicer 1.2.9 its= `; external perimeters extrusion width = 3.00mm `
in the Slicer 1.3.0 its = `; external perimeters extrusion width = 3.00mm (535.62mm^3/s)`

Im using algorithm to get the **3.00** based on the number of the character in the string. 
`Double.Parse(input.Substring(start, ende - start - 2), CultureInfo.InvariantCulture)`

That was the first error that I found. Maybe there are more, but I don't know. I want to use just the stable version, because.... its stable :D

What is a source checkout of the tag? And how do I get that?
Sorry if my lack of knowledge disturbs you.

lordofhyphens, 2017-08-25:
1.3.0-dev has fewer bugs than 1.2.9.

1.2.9 does not get changed at all (all fixes go to 1.3.0-dev). You should
also confirm that your post processor is still necessary ;)

lordofhyphens, 2017-08-25:
As to check out  a tag, that is a general Git question. The tag is 1.2.9,
and I refer you the Internet at large, for that has been answered better
before:

All links should get you to the same place. If you don't understand one
author, move onto the next.

https://stackoverflow.com/questions/35979642/how-to-checkout-remote-git-tag
https://stackoverflow.com/questions/791959/download-a-specific-tag-with-git
http://pinkstone.co.uk/how-to-checkout-a-previous-tag-in-git/
https://git-scm.com/docs/git-clone

stepkurniawan, 2017-08-25:
1. Ah so Tag is basically the version number. Cool! 

2. So... the developer version is more stable than the stable version? :D

3. Yes, as you said, all fixes go to 1.3.0. That's why the outcome of the gcode file is different. I don't see that as a bug, its not even a gcode flavor, its just the comment flavor. 

4. I downloaded (checked-out) the 1.2.9 tag, and tried to find the line for the Wipe class again, but I couldn't find it. And that's my problem is.. I just don't know where to find it.

lordofhyphens, 2017-08-25:
That means it is probably in Perl as of 1.2.9.

There really is no stable version. There are just versions that don't get
any work on them. There was a plan at one point to have a stable branch but
that died due to lack of resources.

stepkurniawan, 2017-08-25:
Owh.. Okay.. sad
Any recomendation on how I should proceed? 
I am not familiar (at all) with Perl, so I would just change my post processor I'd say.

lordofhyphens, 2017-08-25:
Yeah, I can agree with that (changing the post processor).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

stepkurniawan, 2017-10-04:
I tested the _scale(10), its working as you said! Thanks!

closed 2017-10-04 by stepkurniawan