slic3r/Slic3r #4093: outer perimeters not sticked together

azalanono, 2017-08-22:
### Version
1.2.9

### Operating system type + version
openSuSE 13.2 x64

### Behavior
Whatever printjob I generate, the printing is more or less the same: While 100% filled areas are printing perfectly (-> so extrusion quantity is correct), the outer perimeters are printing in a little too wide distance. As result, they are not sticking together and can be peeled off by scratching over the surface with a finger nail.

It would be very helpful to make either the spacing of the outer perimeters adjustable, or to allow adjustment of extrusion quantity. Both will solve the printing issue.

lordofhyphens, 2017-08-22:
I can't reproduce this on my own 3 printers.

Tinchus2009, 2017-08-29:
First you should try useing any of the new DEV version, 1.2.9 is kind of really old.
Then: haveing a "perfect"infill doesnt mean your extrusion is right yu could be perfectly under or over extrudeing and your infill still would look ok. Your outer perimeter not sticking could be one of these reasons:
You are underextrudeing
Your noozle is clooged (and since the outer perimter usually is 0.4 thick, this is where you are going to notice it)
I  have printed a lot with 1.2.9 and never saw that problem, so Im pretty sure you have a extrusion problem rather than a slice problem

lordofhyphens, 2017-08-29:
Tinchus2009 see #4015

<https://github.com/alexrj/Slic3r/issues/4093#issuecomment-325649457>, or mute
the thread

Tinchus2009, 2017-08-29:
auch... well, time to upgrade ubuntu?

lordofhyphens, 2017-08-29:
@Tinchus2009  I'm trying to resolve those problems myself (and could use help with testing that old distro); but your initial indications (flow/extrusion) line up with my own experiences.

@azalanono I *really* need a configuration and model you are seeing this problem with. If you can't/won't provide that I can't even begin to try to fix this.