slic3r/Slic3r #4095: Feature Request: Automatic Variable Layer Height Calculator / Adaptive Slicing

NateTG, 2017-08-25:
Currently the variable layer height interface calls on the user to manually set the layer heights for the object, but there's a relatively simple way to calculate them if we know how much we want the filament extrusions to overlap.

If we have a max step over s, and a facet whose unit normal has non-zero z-component N_z, then a layer height of no more than s * sqrt(N_z^-2 - 1) will ensure that the step over is less than or equal to s.

Facets with N_z equal to zero will always have zero step-over because they are vertical so they can use some maximum layer height.

We know where the top and bottom of each facet is, so we can calculate an interval and a maximum layer height for each facet, and then merge those intervals to generate a layer height profile for the entire print.

Tinchus2009, 2017-08-26:
Sorry, is there now a variable layer height setting?

NateTG, 2017-08-27:
> Sorry, is there now a variable layer height setting?

There's a manual one that's been around for a while.

http://manual.slic3r.org/expert-mode/variable-layer-height

lordofhyphens, 2017-09-02:
https://github.com/alexrj/Slic3r/pull/1386 Anything to do with this?

lordofhyphens, 2017-09-03:
Also, the GcodeTimeEstimator class is currently being called by the Host/Controller for its estimates.

NateTG, 2017-09-03:
> #1386 Anything to do with this?

Definitely.  I'm not sure why I didn't find it while searching.  This could reasonably be considered a dup.

The referenced thesis is not linked to anymore.

I'm not sure what "GcodeTimeEstimator" has to do with this.

lordofhyphens, 2017-09-03:
Just me derping. #close

closed 2017-09-03 by lordofhyphens

lordofhyphens, 2017-09-29:
@NateTG Also, adaptive slicing is now merged.