slic3r/Slic3r #4097: Restart required when "Send to printer" fails due to slicing

forkineye, 2017-08-26:
### Version
Version 1.3.0-dev
Slic3r-master.2017.08.14.1313.a54ca92.64bit

### Operating system type + version
Windows 10 64-bit

### Behavior
* "Send to printer" and "Export G-Code" buttons do not re-activate if slicing fails when "Send to printer" is clicked.  To reproduce, load a model that fails to slice. For instance, scale a bed test below the minimal layer height.  In order for the buttons to become active, you must restart Slic3r.  Re-loading models or fixing the slicing issue doesn't re-activate the buttons.

lordofhyphens, 2017-09-24:
[4097.zip](https://github.com/alexrj/Slic3r/files/1327818/4097.zip)

With the attached configuration and model, deleting the model and re-adding enabled the button for gcode export and print on Linux. Or adding new models.

lordofhyphens, 2017-10-02:
@forkineye can you re-check with a recent -dev build?

forkineye, 2020-11-20:
Closing out old issues

closed 2020-11-20 by forkineye