slic3r/Slic3r #4098: Brim not created around small nested objects

Tinchus2009, 2017-08-26:
### Version
latest dev and previous ones too

### Operating system type + version
win10 64b

### Behavior
 If you have a full palte and need space sometimes you arrange small objects inside biger ones. The problem now arises because in you use a brim, those smaller objeect dont get one.

In the picture you can see an example

[image]

lordofhyphens, 2017-08-27:
Have you set an interior brim?

Also, could you submit an AMF and config you see this problem with?

Tinchus2009, 2017-08-27:
Yes, I tried with interior brim, wich makes a brim in the interior of all objects, included the small nested one, but still thoses nested objects dont get an exterior brim. Config file and amf file atached
[falla_brim.zip](https://github.com/alexrj/Slic3r/files/1254105/falla_brim.zip)

lordofhyphens, 2017-09-03:
My suspicion is that the nested parts are being masked by the one they are inside, because the outer brim logic only looks at the outer contours when making brims. The holes are available too.

One of the things we could do is make brims symmetrical (I suspect this should have been the case in the first place), but that tends to generate brims everywhere (including small holes).

lordofhyphens, 2017-09-03:
[image]
Because that's what happens if I make the brim symmetrical immediately. The interior brim area becomes the max of the interior brim configuration and the exterior.

lordofhyphens, 2017-09-03:
[4098.zip](https://github.com/alexrj/Slic3r/files/1272916/4098.zip)
Here's my test case, simplified slightly, from the original plate.

Ideally, with the provided settings there should be no brim generated inside the holes for the same part and nothing is overlapping.

Tinchus2009, 2017-09-15:
I downloaded your zip. Then loaded the amf file, then loaded your config.ini, the brim is still not being generated for the nsted object