slic3r/Slic3r #4099: Octopi upload fails with Mac OS Parental controls enabled

brownjd, 2017-08-28:
### Version
1.29

### Operating system type + version
Mac OS X 10.11

### Behavior
Octoprint Test fails when Slic3r is launched using a restricted or managed account without full admin access. If I switch users on the same machine to an admin account, Slic3r is able to successfully connect to Octopi instance. The non-admin user is explicitly allowed to use Slic3r (in parental controls panel) and Slic3r is allowed to make and received connections in firewall settings.

When the test button is clicked, the only error generated is from Slic3r indicating the user should check settings and versions. Normally, if an app requires additional rights, Mac OS will prompt for an admin user/passwortd to allow the app to proceed. This does not happen, so it seems the issue is related to the way the connection is initiated in Slic3r (perhaps in a non-standard way that doesn't trigger the OS to requires permission).

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs

lordofhyphens, 2017-08-28:
The connection is initiated through basically cURL (outgoing web
connection). Do you need to whitelist the printer in the parental controls?

See if you can connect to the printer using a web browser on the affected
account.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

brownjd, 2017-08-28:
Is cURL externally referenced? I might need to whitelist it. Do you know the path to the executable?

The printer URL is accessible from the web browser.

Jason Brown

lordofhyphens, 2017-08-28:
it's the perl exec, but I don't know how that interacts (at all) with how
the packaging is (inside the application).

@alexrj might know more (I am very much not a Mac person).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

fessyfoo, 2017-08-29:
I'm pretty sure perl is making the call to octopi with the LWP perl library.  On my Mac the script is in this directory: 
`/Applications/slic3r.app/Contents/MacOS/slic3r.pl `    (adjust for your path to Slic3r)    but that's a script being executed by `/usr/bin/perl` (actually `/usr/bin/env perl`  but likely /usr/bin/perl). 

not sure if that helps you figure out what to whitelist for parental controls.

lordofhyphens, 2017-08-29:
Odd, it should be running perl-local, not system perl.

Does it work if you white list the system perl (as a test)?

brownjd, 2017-08-29:
I’ll check tonight. What command should I execute in order to test perl’s connectivity?

Jason Brown

fessyfoo, 2017-08-29:
correction, is indeed running the perl-local,   so  that would be the target binary to whitelist for parental controls. 

this is it from ps on my machine: 
`/Applications/slic3r-1.3.0-dev-13f121e.app/Contents/MacOS/perl-local -I/Applications/slic3r-1.3.0-dev-13f121e.app/Contents/MacOS/local-lib/lib/perl5 /Applications/slic3r-1.3.0-dev-13f121e.app/Contents/MacOS/slic3r.pl`

brownjd, 2017-08-29:
I don’t see a perl-local executable in that directory.

Also, I’m using 1.2.9 if that makes a difference.

ls -lah /Applications/Slic3r.app/Contents/MacOS/
total 968
drwxr-xr-x@ 3 kids  admin   102B Jun  3 20:57 .
drwxr-xr-x@ 8 kids  admin   272B Jun  3 20:57 ..
-rwxrwxr-x@ 1 kids  admin   482K Jun  3 20:57 slic3r

Jason Brown

lordofhyphens, 2017-08-29:
Yes that makes a huge difference. Try with 1.3.0-dev

brownjd, 2017-08-29:
Well, okay, but will 1.2.9 not work?

Jason Brown

lordofhyphens, 2017-08-30:
They are packaged differently. Also, 1.2.9 isn't being fixed.

brownjd, 2017-08-30:
Yes, but do you know how the connection is invoked in 1.2.9? I’d like to see if I can whitelist it.

Jason Brown

lordofhyphens, 2017-08-31:
The Perl code shouldn't have changed enough from 1.2.9, but the packaging is completely different. I don't know what black magic the 1.2.9 packaging did (it was made with Cava packager).

brownjd, 2017-09-01:
I tested it with 1.3.0 and it correctly asked me to authorize perl-local after I started the app. It seems to connect just fine, so I bet the issue is with authorizing the perl executable in 1.2.9. What’s weird is that the user is never prompted to authorize it.

Anyway, thanks for the help. Might put a FAQ entry if you can so folks know to authorize it. I expect a lot of parents have kids who use Slic3r.

As an aside, 1.3.0 is FAST. Small feature request: I’d love if there were buttons on the toolbar or keyboard shortcuts for rotating and flipping objects. Labeling the axes would also be nice. I know they’re colored, but I can’t ever remember which is which.

Jason Brown