slic3r/Slic3r #4101: Negative XY compensation apparently not working

pierbiag, 2017-08-28:
### Version
1.3.0dev

### Operating system type + version
Win10

### Behavior
Trying to print a simple calibration part, but happens with multiple parts.
If i set the XY compensation value to a positive number (e.g. 1), the inner holes correctly shrink, but if i set a negative one (e.g -1), the result seems identical to the 0 case.

#### STL/Config (.ZIP) where problem occurs
https://www.thingiverse.com/thing:916214

pierbiag, 2017-08-29:
When trying on 1.2.9 it works OK, so it's a current version problem.

lordofhyphens, 2017-08-31:
@pierbiag Could you post a config as well so we can see a full non-working case?

lordofhyphens, 2017-09-03:
Fixed.  Good catch @pierbiag 
[4101.zip](https://github.com/alexrj/Slic3r/files/1272682/4101.zip)

Manual test: 
With provided archive, load stl and generate gcode for different xy-size-compensation values from command line; diff gcode generated with 0 (and the same magnitude positive number) to ensure that the gcode generated is not identical. 
`slic3r.pl --load config.ini --xy-size-compensation=1 euroXY_v1.stl`

closed 2017-09-03 by lordofhyphens

pierbiag, 2017-09-03:
Thank you very much for your help and sorry for not replying earlier. I won't be able to try it for a few days.

lordofhyphens, 2017-09-03:
I was able to verify the output gcode with your test case. The perimeter
shift is easy to spot.

commit c84bc713 (2017-09-03): Regression: Check on absolute value of xy_size_compensation instead of raw value > 0 to permit negative offsets again.
Fixes #4101