slic3r/Slic3r #4102: feature request: support nudge in 2d plater view.

fessyfoo, 2017-08-29:
### Version

slic3r-1.3.0-dev-13f121e

### Operating system type + version

OSX  10.11.6 

### Feature Request: 

in the plater view, in 2d mode.   I would like to use the arrow keys to move the selected object 1 mm at a time.  (or configurable amount)  ie,  nudge.  

my bed is small and i'm often carefully aligning a set of parts.  this feature would make  that much easier. 

thanks for your work and your consideration.

lordofhyphens, 2017-08-31:
@Samir55 you want to take a swing at this when you're done with the other PR?

Samir55, 2017-08-31:
@lordofhyphens I have already started working on it [in this branch ](https://github.com/Samir55/Slic3r/commits/2d_platter_nudge) but I haven't finished it yet :)

lordofhyphens, 2017-08-31:
Open up a PR and reference this so I can remember not to work on it myself. ;)

Samir55, 2017-08-31:
OK :) :)

closed 2018-01-06 by lordofhyphens

fessyfoo, 2018-01-08:
whoo hoo!  thanks.  this works really well!

PR #4104 (2018-01-06): Support moving objects using keyboard arrows in the 2d plater view.
_**Fixes #4102.**_